cortex-lab/phy #1260: zsh: segmentation fault phy template-gui params.py (Mac M1, MacOS 13.4.1 (22F82))

dingyun321, 2024-03-16:
Hi,

I am using a M1 Macbook, the current OS version is 13.4.1.
I fail to open phy GUI and get this error: "zsh: segmentation fault  phy template-gui params.py”.

As I read the previous issue, I get the information by running "**conda list**":
# packages in environment at /opt/anaconda3/envs/phy2:
#
# Name                    Version                   Build  Channel
abseil-cpp                20230802.0           h313beb8_2  
appnope                   0.1.2           py311hca03da5_1001  
arrow-cpp                 14.0.2               hc7aafb3_1  
asttokens                 2.0.5              pyhd3eb1b0_0  
aws-c-auth                0.6.19               h80987f9_0  
aws-c-cal                 0.5.20               h80987f9_0  
aws-c-common              0.8.5                h80987f9_0  
aws-c-compression         0.2.16               h80987f9_0  
aws-c-event-stream        0.2.15               h313beb8_0  
aws-c-http                0.6.25               h80987f9_0  
aws-c-io                  0.13.10              h80987f9_0  
aws-c-mqtt                0.7.13               h80987f9_0  
aws-c-s3                  0.1.51               h80987f9_0  
aws-c-sdkutils            0.1.6                h80987f9_0  
aws-checksums             0.1.13               h80987f9_0  
aws-crt-cpp               0.18.16              h313beb8_0  
aws-sdk-cpp               1.10.55              h313beb8_0  
blas                      1.0                    openblas  
bokeh                     3.3.4           py311hb6e6a13_0  
boost-cpp                 1.82.0               h48ca7d4_2  
bottleneck                1.3.7           py311hb9f6ed7_0  
brotli                    1.0.9                h1a28f6b_7  
brotli-bin                1.0.9                h1a28f6b_7  
brotli-python             1.0.9           py311h313beb8_7  
bzip2                     1.0.8                h80987f9_5  
c-ares                    1.19.1               h80987f9_0  
ca-certificates           2023.12.12           hca03da5_0  
certifi                   2024.2.2        py311hca03da5_0  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
click                     8.1.7           py311hca03da5_0  
cloudpickle               2.2.1           py311hca03da5_0  
colorcet                  3.1.0                    pypi_0    pypi
comm                      0.2.1           py311hca03da5_0  
contourpy                 1.2.0           py311h48ca7d4_0  
cycler                    0.11.0             pyhd3eb1b0_0  
cyrus-sasl                2.1.28               h9131b1a_1  
cython                    3.0.8           py311h80987f9_0  
cytoolz                   0.12.2          py311h80987f9_0  
dask                      2023.11.0       py311hca03da5_0  
dask-core                 2023.11.0       py311hca03da5_0  
debugpy                   1.6.7           py311h313beb8_0  
decorator                 5.1.1              pyhd3eb1b0_0  
distributed               2023.11.0       py311hca03da5_0  
executing                 0.8.3              pyhd3eb1b0_0  
expat                     2.5.0                h313beb8_0  
fonttools                 4.25.0             pyhd3eb1b0_0  
freetype                  2.12.1               h1192e45_0  
fsspec                    2023.10.0       py311hca03da5_0  
gettext                   0.21.0               h13f89a0_1  
gflags                    2.2.2                h313beb8_1  
ghp-import                2.1.0                    pypi_0    pypi
glib                      2.78.4               h313beb8_0  
glib-tools                2.78.4               h313beb8_0  
glog                      0.5.0                h313beb8_1  
grpc-cpp                  1.48.2               hc60591f_4  
gst-plugins-base          1.14.1               h313beb8_1  
gstreamer                 1.14.1               h80987f9_1  
gtest                     1.14.0               h48ca7d4_0  
h5py                      3.9.0           py311hba6ad2f_0  
hdf5                      1.12.1               h05c076b_3  
heapdict                  1.0.1              pyhd3eb1b0_0  
[... 169 lines cut ...]
20:33:50.110 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
20:33:50.110 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
20:33:50.110 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
20:33:50.111 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
20:33:50.111 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
20:33:50.111 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
20:33:50.111 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
20:33:50.111 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
20:33:50.112 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
20:33:50.112 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
20:33:50.115 [D] context:185          Save data to `/Users/yunding/Downloads/Pen4a/ks2/sorter_output/.phy/spikes_per_cluster.pkl`.
20:33:50.119 [D] gui:463              Creating GUI.
20:33:50.169 [D] state:46             Load /Users/yunding/.phy/TemplateGUI/state.json for GUIState.
20:33:50.170 [D] state:46             Load /Users/yunding/Downloads/Pen4a/ks2/sorter_output/.phy/state.json for GUIState.
20:33:50.317 [D] gui:718              Add view ClusterView to GUI.
20:33:50.346 [D] gui:718              Add view SimilarityView to GUI.
20:33:50.369 [D] gui:718              Add view WaveformView to GUI.
20:33:50.394 [D] base:337             Set state for WaveformView.
20:33:50.401 [D] gui:718              Add view CorrelogramView to GUI.
20:33:50.403 [D] base:337             Set state for CorrelogramView.
20:33:50.414 [D] gui:718              Add view ISIView to GUI.
20:33:50.416 [D] base:337             Set state for ISIView.

I sincerely appreciate your feedback!
Yun

zm711, 2024-03-17:
@dingyun321, Unfortunately under the hood Phy relies on openGL. Since Phy is in maintenance mode and not in active development I doubt this will be fixed for you (ie Apple deprecated openGL support many versions ago in Macs).

dingyun321, 2024-03-18:
I see. Do you know the last system version on a Mac that can run Phy?

Thank you very much!

louiskang, 2024-04-22:
I'm using an Intel Mac with MacOS 14.3.1, and somehow this installation worked for me. I use Intel MKL as my BLAS library, so maybe it would work for you if you remove `"libblas=*=*mkl"`?

```
conda create -n phy python=3.9 numpy=1.23 "libblas=*=*mkl"
conda activate phy
pip install phy --pre --upgrade
pip uninstall pyqt5 pyqtwebengine
pip install pyqt5==5.14.2 pyqtwebengine==5.14.0
```

closed 2024-06-04 by zm711

YufeiSi, 2024-09-06:
> I'm using an Intel Mac with MacOS 14.3.1, and somehow this installation worked for me. I use Intel MKL as my BLAS library, so maybe it would work for you if you remove `"libblas=*=*mkl"`?
>
> ```
> conda [...]

This helped my Mac (MacOS 13.0.1, Intel) when removed "libblas=*=*mkl". I wasn't able to install the versions listed here for pyqt5 and pyqtwebengine on my M1 Mac with MacOS 14.4.1 though.