cortex-lab/phy #1262: Shank visualization

MFalappa, 2024-03-25:
Hello, 
I am an engineer at Corticale and I am using Phy on some data collected with our probes. 

Currently, I am analyzing data from a probe with 8 shanks (2 columns and 64 rows for each shank, for a total of 1024 channels). if I have understood correctly, it seems  that traces from the same shank have the same color, but they are not contiguous as they are organized by depth. 
I was wondering if it would be possible to organize the channels in TraceView not only by depth but also by index, or at least to organize the traces so that those belonging to the same shank are contiguous.

thanks in advances

[image]

MFalappa, 2024-03-25:
In the meantime, in TraceView class, I changed self.channel_y_ranks = np.argsort(np.argsort(self.channel_positions[:, 1])) to self.channel_y_ranks = np.arange(self.channel_positions.shape[0], dtype='int64')[::-1], but it would be good if it is possible switching between the two visualization in the gui

zm711, 2024-04-12:
They should. If a channel is next to another channel on the same shank phy will automatically display them next to each other. If it is not there is one of the two things
1) confusion about the channel contact vs the data number (ie an electrode might be number 1, but might come from data row 10 so phy will list it as 10 not 1)
2) the channel map was made wrong.