cortex-lab/phy #1264: AmplitudeView not showing raw due to error "model:667 Skipping spike waveforms"

K-Onod, 2024-03-27:
I set up phy2 on my new Windows 11. I successfully loaded up phy2 (phy TemplateGUI v2.0b5, phylib v2.4.3) using environment.yml ([#1218](https://github.com/cortex-lab/phy/issues/1218)). After launching phy template-gui params.py, I can open phy, but then encountered:

17:46:01.340 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
17:46:01.640 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
17:46:01.790 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.

I think that all phy operations would work, except for AmlitudeView not showing raw amplitude, which can be explained by model:667 sentence. I successfully ran phy using the same spike data on my old computer where phy2 was loaded more than three years ago (phy TemplateGUI v2.0a1, phylib v2.0a3). So, the error reason does not result from the data itself.

How can I fix it?

Thanks for your help!

--
PS. I attached the conda lists and "phy template-gui params.py --debug" outputs below.

packages in environment at C:\Users\***\anaconda3\envs\phy2:

Name                    Version                   Build  Channel
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
aws-c-auth                0.7.16               h7613915_8    conda-forge
aws-c-cal                 0.6.10               hf6fcf4e_2    conda-forge
aws-c-common              0.9.14               hcfcfb64_0    conda-forge
aws-c-compression         0.2.18               hf6fcf4e_2    conda-forge
aws-c-event-stream        0.4.2                h3df98b0_6    conda-forge
aws-c-http                0.8.1                h4e3df0f_7    conda-forge
aws-c-io                  0.14.6               hf0b8b6f_2    conda-forge
aws-c-mqtt                0.10.3               h96fac68_2    conda-forge
aws-c-s3                  0.5.4                h08df315_0    conda-forge
aws-c-sdkutils            0.1.15               hf6fcf4e_2    conda-forge
aws-checksums             0.1.18               hf6fcf4e_2    conda-forge
aws-crt-cpp               0.26.4               hbe739fa_2    conda-forge
aws-sdk-cpp               1.11.267             hfaf0dd0_4    conda-forge
bokeh                     3.4.0              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
brotli-python             1.1.0           py310h00ffb61_1    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-ares                    1.27.0               hcfcfb64_0    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
click                     8.1.7           win_pyh7428d3b_0    conda-forge
cloudpickle               3.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
colorcet                  3.1.0                    pypi_0    pypi
comm                      0.2.2              pyhd8ed1ab_0    conda-forge
contourpy                 1.2.0           py310h232114e_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
cython                    3.0.9           py310h00ffb61_0    conda-forge
cytoolz                   0.12.3          py310h8d17308_0    conda-forge
dask                      2024.3.1           pyhd8ed1ab_0    conda-forge
dask-core                 2024.3.1           pyhd8ed1ab_0    conda-forge
dask-expr                 1.0.5              pyhd8ed1ab_0    conda-forge
debugpy                   1.8.1           py310h00ffb61_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
distributed               2024.3.1           pyhd8ed1ab_0    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
fonttools                 4.50.0          py310h8d17308_0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
fsspec                    2024.3.1           pyhca7485f_0    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
ghp-import                2.1.0                    pypi_0    pypi
git                       2.44.0               h57928b3_0    conda-forge
glib                      2.80.0               h39d0aa6_1    conda-forge
glib-tools                2.80.0               h0a98069_1    conda-forge
gst-plugins-base          1.22.9               h001b923_1    conda-forge
gstreamer                 1.22.9               hb4038d2_1    conda-forge
[... 219 lines cut ...]
18:27:01.347 [D] base:337             Set state for WaveformView.
18:27:01.355 [D] gui:718              Add view CorrelogramView to GUI.
18:27:01.357 [D] base:337             Set state for CorrelogramView.
18:27:01.371 [D] gui:718              Add view ISIView to GUI.
18:27:01.373 [D] base:337             Set state for ISIView.
18:27:01.383 [D] gui:718              Add view FeatureView to GUI.
18:27:01.385 [D] base:337             Set state for FeatureView.
18:27:01.401 [D] gui:718              Add view AmplitudeView to GUI.
18:27:01.402 [D] base:337             Set state for AmplitudeView.
18:27:01.416 [D] gui:718              Add view FiringRateView to GUI.
18:27:01.418 [D] base:337             Set state for FiringRateView.
18:27:01.542 [D] gui:718              Add view TraceView to GUI.
18:27:01.544 [D] base:337             Set state for TraceView.
18:27:01.544 [D] base:494             Set color scheme to random.
18:27:01.656 [D] base:1626            Adding default color schemes to TraceView.
18:27:01.693 [D] gui:718              Add view ProbeView to GUI.
18:27:01.695 [D] base:337             Set state for ProbeView.
18:27:01.707 [D] gui:718              Add view TemplateFeatureView to GUI.
18:27:01.708 [D] base:337             Set state for TemplateFeatureView.
18:27:01.723 [D] gui:718              Add view ClusterScatterView to GUI.
18:27:01.724 [D] base:337             Set state for ClusterScatterView.
18:27:01.724 [D] base:494             Set color scheme to random.
18:27:01.725 [D] base:1626            Adding default color schemes to ClusterScatterView.
18:27:01.815 [D] gui:519              Load the geometry state.
18:27:02.283 [D] supervisor:817       Clusters selected: [235] (197)

alirza67, 2024-04-08:
> I set up phy2 on my new Windows 11. I successfully loaded up phy2 (phy TemplateGUI v2.0b5, phylib v2.4.3) using environment.yml ([#1218](https://github.com/cortex-lab/phy/issues/1218)). After launching phy template-gui params.py, I can open phy, but then encountered:
> 
> 17:46:01.340 [W] model:603 Unreferenced clusters found in templates (generally not a problem) 17:46:01.640 [W] model:625 Unreferenced clusters found in spike_clusters (generally not a problem) 17:46:01.790 [W] model:667 Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 
> I think that all phy operations would work, except for AmlitudeView not showing raw amplitude, which can be explained by model:667 sentence. I successfully ran phy using the same spike data on my old computer where phy2 was loaded more than three years ago (phy TemplateGUI v2.0a1, phylib v2.0a3). So, the error reason does not result from the data itself.
> 
> How can I fix it?
> 
> Thanks for your help!
> 
> -- PS. I attached the conda lists and "phy template-gui params.py --debug" outputs below. [text1.txt](https://github.com/cortex-lab/phy/files/14833516/text1.txt) [text2.txt](https://github.com/cortex-lab/phy/files/14833518/text2.txt)

Hi, 
I am having the same problem. I don't think the not having the amplitude is due to "model:667 Skipping spike waveforms" .
you can solve the "skipping spike waveform" by the following command:
 phy  extract-waveforms params.py

K-Onod, 2024-04-09:
Thanks, alirza67! The "[W] model:667 Skipping spike waveforms" was gone after launching phy extract-waveforms params.py. I appreciate you.

Yet, it has no raw amplitude for AmplitudeView. That raw amplitude is very useful for removing sudden artifacts related to motion or LED illumination. So, I still need to fix it.

zm711, 2024-04-12:
@K-Onod,

Which KS are you using? I agree amplitudes are super important :) based on your --debug it looks like Phy thinks it's loading amplitudes. So I think we need to start with what sorter you used and are there amplitude files in the directory?

K-Onod, 2024-04-13:
Hi, zm711. Here is my dataset loaded in phy. Please let me know if you need actual files.
[image]

zm711, 2024-04-13:
And which kilosort are you using?

K-Onod, 2024-04-15:
I use kilosort 2.5.

K-Onod, 2024-12-18:
That problem was fixed. See https://github.com/cortex-lab/phy/issues/1175.

closed 2024-12-18 by K-Onod