cortex-lab/phy #1265: Unable to switch to waveform type outside of templates

danielpollak, 2024-03-29:
When I click "w" or select "Next Waveform Type" in the WaveformView, it does not change. I ran 

`phy template-gui params.py --debug`

and the log when I do these actions is: 
"Switch to waveforms type templates."

I installed the latest version of phy on Windows 10 using the following commands:
`conda create -n phy2 -y cython dask h5py joblib matplotlib numpy pillow pip pyopengl pyqt pyqtwebengine pytest python qtconsole requests responses scikit-learn scipy traitlets python=3.9`

`conda activate phy2`

`pip install git+https://github.com/cortex-lab/phy.git`

Any ideas what I can do?

zm711, 2024-04-12:
I actually advise people to install using the environment.yml to get the updated version.  Only issue is you have to edit the line that says
```
python
```
to be 
```
python=3.11
```
because 3.12 removed the `imp` module which will break phy. 

Does that sound reasonable?

danielpollak, 2024-04-12:
Hey @zm711, thanks for taking a look. Unfortunately that didn't work, but I was able to install a previous version of phy, which had this functionality. I think it depends in part on the version of KS that sorted the data. I used KS 2.5 or 3 (not sure which), but when I re-sorted using KS4.0, the latest version of phy, installed the way it suggests in the docs, works great. Not sure why older versions of KS and current versions of phy aren't compatible, but I got around it.

danielpollak, 2024-04-12:
For anyone wondering how to install an older version of phy, here are my notes on the installation process. Idk why it's pasting as an image instead of text. Sorry for inconvenience.

[image]

closed 2024-04-12 by danielpollak

zm711, 2024-04-12:
I'm curious what the problem is. Current phy works fine with KS2, 2.5, and 3 (I just installed it on Windows 10 and Windows 11 within the past couple months). KS4 still has bugs so I wouldn't blame phy if it doesn't always work with KS4 (they need to fix a lot of bugs right now). Ideally you wouldn't use 3.7 since it's no longer supported, but I guess we have to use what works :) 

(I also followed those old instructions before the most recent update last June :) )