cortex-lab/phy #1271: Segmentation Fault when running phy

mattslab, 2024-04-25:
I get a Segmentation Fault when running phy. This is Ubuntu 22.04, with phy installed via anaconda. Had to force 3.11 install, which I did using this:

$ conda create --name phy_custom python=3.11
$ conda install cython dask h5py joblib matplotlib numpy pillow pyopengl pyqt pyqtwebengine pytest qtconsole requests responses scikit-learn scipy traitlets --freeze-installed

Then follow the rest of the readme instructions. Doing this got around the "imp" issue that I had at first, and phy runs. However, I get a seg fault. More info below on the debug output and the conda environment, but any ideas? Thanks!

(phy_custom) $ phy template-gui params.py
08:54:08.740 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
Segmentation fault (core dumped)

running with debug gives this:
(phy_custom) $ phy template-gui params.py --debug
08:54:46.075 [D] __init__:68          Start capturing exceptions.
08:54:46.650 [D] model:619            Loading spike clusters.
08:54:47.144 [D] model:569            No channel shank file found.
08:54:47.144 [D] model:692            Loading templates.
08:54:47.155 [D] model:724            Templates are dense.
08:54:47.161 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
08:54:47.161 [D] model:730            Loading the whitening matrix.
08:54:47.161 [D] model:737            Loading the inverse of the whitening matrix.
08:54:47.163 [D] model:766            Loading features.
08:54:47.163 [D] model:781            Features are sparse.
08:54:47.163 [D] model:803            Loading template features.
08:54:47.171 [D] model:530            Load spike_detection_templates.npy.
08:54:47.199 [D] model:530            Load spike_positions.npy.
08:54:47.200 [D] model:504            Load `cluster_ContamPct.tsv`.
08:54:47.201 [D] model:504            Load `cluster_KSLabel.tsv`.
08:54:47.203 [D] model:504            Load `cluster_group.tsv`.
08:54:47.204 [D] model:504            Load `cluster_Amplitude.tsv`.
08:54:47.467 [D] context:100          Initialize joblib cache dir at `/media/neuropixels/Ya_240415_s2190003_g0/Ya_240415_s219_0003_g0_imec0/kilosort4/.phy`.
08:54:47.467 [D] context:101          Reducing the size of the cache if needed.
08:54:47.469 [D] base:102             Add filter `high_pass`.
08:54:47.469 [D] config:31            Load config file `/home/smithlab/.phy/phy_config.py`.
08:54:47.470 [D] plugin:145           Loading 0 plugins.
08:54:47.471 [D] context:209          The file `/media/neuropixels/Ya_240415_s219_0003_g0/Ya_240415_s219_0003_g0_imec0/kilosort4/.phy/new_cluster_id.pkl` doesn't exist.
08:54:47.596 [D] context:185          Save data to `/media/neuropixels/Ya_240415_s219_0003_g0/Ya_240415_s219_0003_g0_imec0/kilosort4/.phy/spikes_per_cluster.pkl`.
08:54:47.754 [D] gui:463              Creating GUI.
08:54:47.760 [D] state:46             Load /home/smithlab/.phy/TemplateGUI/state.json for GUIState.
08:54:48.689 [D] gui:718              Add view ClusterView to GUI.
08:54:48.694 [D] gui:718              Add view SimilarityView to GUI.
Segmentation fault (core dumped)

And conda list gives this:
(phy_custom) $ conda list
# packages in environment at /home/username/anaconda3/envs/phy_custom:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
abseil-cpp                20211102.0           hd4dd3e8_0  
arrow-cpp                 14.0.2               h374c478_1  
asttokens                 2.0.5              pyhd3eb1b0_0  
aws-c-auth                0.6.19               h5eee18b_0  
aws-c-cal                 0.5.20               hdbd6064_0  
aws-c-common              0.8.5                h5eee18b_0  
aws-c-compression         0.2.16               h5eee18b_0  
aws-c-event-stream        0.2.15               h6a678d5_0  
aws-c-http                0.6.25               h5eee18b_0  
aws-c-io                  0.13.10              h5eee18b_0  
aws-c-mqtt                0.7.13               h5eee18b_0  
aws-c-s3                  0.1.51               hdbd6064_0  
aws-c-sdkutils            0.1.6                h5eee18b_0  
aws-checksums             0.1.13               h5eee18b_0  
aws-crt-cpp               0.18.16              h6a678d5_0  
aws-sdk-cpp               1.10.55              h721c034_0  
blas                      1.0                         mkl  
bokeh                     3.4.0           py311h92b7b1e_1  
boost-cpp                 1.82.0               hdb19cb5_2  
bottleneck                1.3.7           py311hf4808d0_0  
[... 174 lines cut ...]
sqlite                    3.41.2               h5eee18b_0  
stack_data                0.2.0              pyhd3eb1b0_0  
tbb                       2021.8.0             hdb19cb5_0  
tblib                     1.7.0              pyhd3eb1b0_0  
threadpoolctl             2.2.0              pyh0d69192_0  
tk                        8.6.12               h1ccaba5_0  
toolz                     0.12.0          py311h06a4308_0  
tornado                   6.3.3           py311h5eee18b_0  
tqdm                      4.66.2                   pypi_0    pypi
traitlets                 5.7.1           py311h06a4308_0  
tzdata                    2024a                h04d1e81_0  
unicodedata2              15.1.0          py311h5eee18b_0  
urllib3                   2.1.0           py311h06a4308_1  
utf8proc                  2.6.1                h5eee18b_1  
watchdog                  4.0.0                    pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0  
wheel                     0.41.2          py311h06a4308_0  
xyzservices               2022.9.0        py311h06a4308_1  
xz                        5.4.6                h5eee18b_0  
yaml                      0.2.5                h7b6447c_0  
zeromq                    4.3.5                h6a678d5_0  
zict                      3.0.0           py311h06a4308_0  
zipp                      3.17.0          py311h06a4308_0  
zlib                      1.2.13               h5eee18b_0  
zstd                      1.5.5                hc292b87_0

zm711, 2024-04-26:
Could you try installing with the `environment.yaml`. But as you noted you need to edit it so that `python=3.11`?

mattslab, 2024-04-26:
> Could you try installing with the `environment.yaml`. But as you noted you need to edit it so that `python=3.11`?

This worked - thank you!

zm711, 2024-06-06:
Cleaning up fixed issues. Feel free to open new issues if needed :)

closed 2024-06-06 by zm711