cortex-lab/phy #1272: phy does not find module imp

ucsfidl, 2024-04-26:
I installed phy on an Ubuntu 22.04 LTS machine with a fresh perfectly working version of Anaconda exactly as specified on the github page (in italics below) and got the error at the bottom.  **Phy** could not find the module **imp.py**   Earlier version of **phy** installed on other machines were fine.

_conda create -n phy2 -y cython dask h5py joblib matplotlib numpy pillow pip pyopengl pyqt pyqtwebengine pytest python qtconsole requests responses scikit-learn scipy traitlets
conda activate phy2
pip install git+https://github.com/cortex-lab/phy.git
cd mykilosort_output_folder
phy template-gui params.py_

(phy2) idl@mps-pc59:/nas/data/2019/2019_06_25-ks$ phy template-gui params.py
Traceback (most recent call last):
  File "/home/idl/yes/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/home/idl/yes/envs/phy2/lib/python3.12/site-packages/phy/__init__.py", line 21, in <module>
    from .utils.config import load_master_config
  File "/home/idl/yes/envs/phy2/lib/python3.12/site-packages/phy/utils/__init__.py", line 8, in <module>
    from .plugin import IPlugin, attach_plugins
  File "/home/idl/yes/envs/phy2/lib/python3.12/site-packages/phy/utils/plugin.py", line 14, in <module>
    import imp
ModuleNotFoundError: No module named 'imp'

zm711, 2024-04-29:
imp was removed from python 3.12. You need to specify <3.12. My advice would be to download the `environment.yaml` and then edit the line with python and set it to 
```
python=3.11
```

zm711, 2024-06-03:
Should be fixed now.

closed 2024-06-03 by zm711