cortex-lab/phy #1273: Error "QWidget: Must construct a QApplication before a QWidget"

adam-hockley, 2024-04-30:
I keep getting this error when trying to open a fresh install of Phy on a dataset sorted with KS4 using spike interface. 
I followed the installation instructions on the readme, with the required changes to the python version "python=3.11" and also I had to run "conda install git".

Any help much appreciated!

Here is the traceback and the conda list:

----------

(phy2_test) C:\SortingFolder\NPH5\5\Sorting\KS4\sorter_output>phy template-gui params.py

10:08:09.916 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
10:08:09.916 [E] __init__:62          An error has occurred (AssertionError): No extension found in file `C:\SortingFolder`
Traceback (most recent call last):
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui
    model = load_model(params_path)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phylib\io\model.py", line 1433, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phylib\io\model.py", line 339, in __init__
    self._load_data()
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phylib\io\model.py", line 449, in _load_data
    self.traces = self._load_traces(self.channel_mapping)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phylib\io\model.py", line 580, in _load_traces
    traces = get_ephys_reader(
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phylib\io\traces.py", line 496, in get_ephys_reader
    klass, arg, kwargs = _get_ephys_constructor(obj, **kwargs)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phylib\io\traces.py", line 483, in _get_ephys_constructor
    klass, arg, kwargs = _get_ephys_constructor(obj[0], **kwargs)
  File "C:\Users\ANL\anaconda3\envs\phy2_test\lib\site-packages\phylib\io\traces.py", line 466, in _get_ephys_constructor
    assert ext, "No extension found in file `%s`" % path
AssertionError: No extension found in file `C:\SortingFolder`

QWidget: Must construct a QApplication before a QWidget

----------

# packages in environment at C:\Users\ANL\anaconda3\envs\phy2_test:
#
# Name                    Version                   Build  Channel
abseil-cpp                20211102.0           hd77b12b_0
arrow-cpp                 14.0.2               ha81ea56_1
asttokens                 2.0.5              pyhd3eb1b0_0
aws-c-auth                0.6.19               h2bbff1b_0
aws-c-cal                 0.5.20               h2bbff1b_0
aws-c-common              0.8.5                h2bbff1b_0
aws-c-compression         0.2.16               h2bbff1b_0
aws-c-event-stream        0.2.15               hd77b12b_0
aws-c-http                0.6.25               h2bbff1b_0
aws-c-io                  0.13.10              h2bbff1b_0
aws-c-mqtt                0.7.13               h2bbff1b_0
[... 155 lines cut ...]
tblib                     1.7.0              pyhd3eb1b0_0
threadpoolctl             2.2.0              pyh0d69192_0
tk                        8.6.12               h2bbff1b_0
toolz                     0.12.0          py311haa95532_0
tornado                   6.3.3           py311h2bbff1b_0
tqdm                      4.66.2                   pypi_0    pypi
traitlets                 5.7.1           py311haa95532_0
tzdata                    2024a                h04d1e81_0
unicodedata2              15.1.0          py311h2bbff1b_0
urllib3                   2.1.0           py311haa95532_1
utf8proc                  2.6.1                h2bbff1b_1
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
watchdog                  4.0.0                    pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0
wheel                     0.41.2          py311haa95532_0
win_inet_pton             1.1.0           py311haa95532_0
xyzservices               2022.9.0        py311haa95532_1
xz                        5.4.6                h8cc25b3_0
yaml                      0.2.5                he774522_0
zeromq                    4.3.5                hd77b12b_0
zict                      3.0.0           py311haa95532_0
zipp                      3.17.0          py311haa95532_0
zlib                      1.2.13               h8cc25b3_0
zstd                      1.5.5                hd43e919_0

zm711, 2024-04-30:
What file format are you using. I haven't seen that specific cause of the error but it seems like it is struggling to read the binary file. Could you post the contents of the `params.py` file here?

asheff794, 2024-05-01:
I was getting the same error. The problem for me was indeed in the params.py file, the dat_path was referring to the directory where I ran kilosort but not the actual bin file. I was running kilosort4 from spikeinterface so I imagine that's why the param.py file was incorrect. 

I changed the dat_path to point to the actual .bin file and everything works now (see below), hope that solves it for you as well!

**Before**
`dat_path = '/vast/palmer/home.mccleary/ags72/Documents/Ephys/'
n_channels_dat = 32
dtype = 'uint16'
offset = 0
sample_rate = 30000.0
hp_filtered = False
`

**After**
`dat_path = '/vast/palmer/home.mccleary/ags72/Documents/Ephys/Test_data/data.bin'
n_channels_dat = 32
dtype = 'uint16'
offset = 0
sample_rate = 30000.0
hp_filtered = False
`

zm711, 2024-05-01:
That's what I assumed the issue was. We were discussing trying to fix this at the spikeinterface level, but we hadn't decided on that yet since some people don't have a binary file with spikeinterface. But just editing the `params.py` manually should fix it when using the wrapper.

closed 2024-06-04 by zm711