cortex-lab/phy #1276: Regarding the issue of drawing 2D probes.

yyyaaaaaaa, 2024-05-20:
Hi, I wish to display a 2D probe array in certain views, such as the clusterscatter view. I want the x-axis and y-axis to represent the physical coordinates of the probes, but currently, both x and y can only be set to depth. I have checked the available properties, and it seems that there is only one property, depth, and there is no property such as length. However, my probes are set up with x and y coordinates.

These are the properties of my clusters.
{'id': 140, 'ch': '258', 'sh': 0, 'depth': 1785.0, 'fr': 0.6933564452148405, 'amp': 58.766525, 'n_spikes': 208, 'amplitude_cutoff': None, 'channel_group': 0, 'group': 'unsorted', 'isi_violations_count': 0.0, 'isi_violations_ratio': 
0.0, 'presence_ratio': 1.0, 'si_unit_id': 140, 'snr': 5.327821730321634, 'is_masked': False}

zm711, 2024-06-04:
I'm not sure what you mean? The probe map should automatically display? could you re-explain what your problem is?

yyyaaaaaaa, 2024-06-06:
Sure！My probe is a 2D probe with x and y coordinates (depth). I want to display the sorting results from Kilosort4 in the ClusterScatterView view. When setting the two axes of the view, I found that I can only set them to depth (y-coordinate). I can't get the real position of the units in this view. I tried to print out all the properties used by Phy for the clusters, and I found that they only contain the depth (y-coordinate) information of the probe, without the other type of location information (x-coordinate).
[image]
[image]

zm711, 2024-06-06:
And you don't have a probe view? Typically the waveform view and the probe view give you a nice highlighting of the x-y(depth) positions of units. So I'm trying to figure out if this is a KS4 problem. If you ran the same data through KS2,2.5,3 does it display better?

yyyaaaaaaa, 2024-06-06:
I got the probeview view. The background of the second picture in the previous reply is probeview, which shows the real information of the electrode. As for other versions of kilosort, I haven't tried it yet.