cortex-lab/phy #1277: GUI empty

EMerinoM14, 2024-05-23:
Hello everybody! I have been trying to install kilosort (v4) and phy for a few days now. I have already managed to perform the first spike sorting through kilosort, however when I go to open the GUI of phy to see how it has been done, I find it empty, I can only see the clusters performed but nothing else. 

Currently I get this error when executing the command "phy template-gui params.py" :

(phy2) C:\Users\Usuario\Desktop\Matlab\2024-04-03_12-56-50\kilosort4>phy template-gui params.py
QMimeDatabase: Error loading internal MIME data
An error has been encountered at line 1 of <internal MIME data>: Premature end of document..:

It is important to say that in order to get to open the phy2 GUII had to do several things. At the beginning I got the following errors:

(phy2) C:\Users\Usuario\Desktop\Matlab\2024-04-03_12-56-50\kilosort4>phy template-gui params.py
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Usuario\.conda\envs\phy2\Scripts\phy.exe\__main__.py", line 4, in <module>
  File "C:\Users\Usuario\.conda\envs\phy2\Lib\site-packages\phy\__init__.py", line 21, in <module>
    from .utils.config import load_master_config
  File "C:\Users\Usuario\.conda\envs\phy2\Lib\site-packages\phy\utils\__init__.py", line 8, in <module>
    from .plugin import IPlugin, attach_plugins
  File "C:\Users\Usuario\.conda\envs\phy2\Lib\site-packages\phy\utils\plugin.py", line 14, in <module>
    import imp
ModuleNotFoundError: No module named 'imp'

To solve it I had to replace in the source code of phy the use of the module "imp" by "importlib". Specifically I had to modify the plugin.py file. I don't know if by modifying this file I have caused that nothing can be seen in the Phy2 GUI. 

I would really appreciate your help!
[image]

EMerinoM14, 2024-05-24:
The problem is solved in a very easy way, it is not necessary to change anything in the plugin.py file. 

By default when you install phy2 it installs the latest version of python (in my case 3.12). The only thing I did was to go back to an earlier version of python (in the phy environment), specifically 3.9.13.

closed 2024-06-04 by zm711

ChauhanT, 2025-10-01:
Hi, I cannot seem to install phy2 on python 3.12. Is this expected ? Thank you!