cortex-lab/phy #1278: error : phy template-gui params.py

TNR1073, 2024-05-29:
Hi, everyone. 
I will use phy2 for spike sorting. 
I installed phy2 following the instruction. When I opened GUI, the error shown below occured.
I'm a beginner. So I don't know the solution.
Please tell me if you know how to solve this issue.

>phy template-gui params.py
12:51:43.646 [E] __init__:62          An error has occurred (KeyError): 'dtype'
Traceback (most recent call last):
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\ryo_1\anaconda3\envs\test\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui
    model = load_model(params_path)
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\phylib\io\model.py", line 1433, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "C:\Users\ryo_1\anaconda3\envs\test\lib\site-packages\phylib\io\model.py", line 1417, in get_template_params
    params['dtype'] = np.dtype(params['dtype'])
KeyError: 'dtype'

QWidget: Must construct a QApplication before a QWidget

zm711, 2024-06-04:
The error is actually not helpful. But the stack trace is. Why are you trying to import statsmodels? Does your lab have some plugins?

TNR1073, 2024-06-05:
Thank you for responding.
I misunderstood and used a different kind of file.
I can resolved this problem.
Thanks.

closed 2024-06-05 by TNR1073