cortex-lab/phy #1279: [HELP] Asign the waveform of a spike_time

Lorena-J, 2024-06-03:
Hello!! I am analysing some data and I need to get a .txt file containing the neuron ID, the time at which each spike of that neuron occurs and the points that form the concrete waveform of that spike. 

For this I am using phylib. I have managed to obtain for each cluster ID the points that form all the waveforms of all the spikes of that cluster ID using: 

````python
model = load_model( path/to/params.py
cluster_id = int(271) # second argument: cluster ID
waveforms = model.get_cluster_spike_waveforms(cluster_id)
```
I need to know the relationship between the spike time and which points make up its concrete waveform.

Thanks for help!!!

zm711, 2024-06-03:
What do you mean by concrete waveform? I think I understand the rest of your question. Like you want a vector of all voltages?

Lorena-J, 2024-06-03:
Yes, I need a vector of all voltages to create a table similar like this:
ID Neuron | Spike time | voltages 
1                 |    1.341      | 0.4, -3.5, 1.78 ....
1                 |    1.478      | 0,-1,-0.34, 6.79 .....
2                |     2.875     | 0.34, 0, 1-1.86 ......

For example, in this case (the data are invented) if I plot the voltages of the first row, I could see the waveform of the spike that takes place in time 1.341  and belongs to cluster 1

zm711, 2024-06-04:
Sorry I'm still not sure what your problem is exactly. So you have the waveforms, but based on your script you don't have the spike times? So you currently have cluster id and waveform values?

Lorena-J, 2024-06-05:
I'm sorry, I don't think I'm explaining myself well. I have the cluster id and the peak times, so I can see the time of the first peak of a selected cluster, but I don't have the waveform of this particular peak. What I need is to know in which file the voltage points that make up a waveform are stored and to know the relationship between the waveform and the time at which that spike occurs.

zm711, 2024-06-05:
Cool. So in general those waveforms are stored in your raw data file. Often this file is the one that says `.bin` or `.dat` at the end. The peak time is as you called it: the peak. So often what people do is memmap the raw file and then you put in the peak time with a range of either -60:+60 or -80:+80 and get the waveforms yourself. This method has the benefit that you can filter the waveforms for post processing (or apply referencing). I actually like recommending spikeinterface for this which will automatically do all this work for you with a few lines of code.

If you take the waveforms that Phy is calculating they will be much rawer, which might be okay, but might not depending on the types of noise in your recording. Does that make more sense?

Lorena-J, 2024-06-05:
More or less, but with the code I put in the first comment the waveforms variable turns out to be an array with a shape of (19041, 82, 12), which, as far as I understand, the first dimension of that three-dimensional array is the waveforms, each with 82 voltage points in 12 different channels(?). If this is correct, what I need to know is how to relate those waveforms to the time at which they occur. That is, of those 19041 waveforms, if my neuron 1 fires at second 3, which waveform is the one for that particular firing.

zm711, 2024-06-05:
Cool that's easy enough. Those should be your spikes. Which should be monotonic. So if you open the `spike_times.npy` file it should be (19041,1) right? If that is the case then `spike_clusters.npy` will give you the id. So what you would do is:

cluster 0:
id: spike_clusters[0]
times: spike_times[0]
waveforms: waveforms[0,:,:]

Sometimes people only want the peak waveform only which requires a bit more work.

Lorena-J, 2024-07-12:
Great!!! tnaks for all your help!!!