cortex-lab/phy #1283: Phy gui is blank apart from cluster view and similarity view

AbhiSwamiUConn, 2024-06-05:
When I open the gui using the command phy template-gui params.py, the gui opens but apart from what is listed in the title, nothing shows up.  I am on a windows 10 machine with an intel x-series processor.  
[image]

zm711, 2024-06-05:
If you run:

```
phy template-gui params.py --debug
```

What appears?

AbhiSwamiUConn, 2024-06-05:
I installed using a yml file containing the following text:
name: phy2
channels:
  - conda-forge
  - defaults
dependencies:
  - python
  - pip
  - git
  - numpy
  - matplotlib
  - scipy
  - h5py
  - pyqt
  - pyopengl
  - pyqtwebengine
  - pytest
  - qtconsole
  - requests
  - responses
  - traitlets
  - dask
  - cython
  - pillow
  - scikit-learn
  - joblib
  - pip:
    - git+https://github.com/cortex-lab/phy.git

zm711, 2024-06-05:
could you do 
```
conda list
```

inside of the environment you created ?

AbhiSwamiUConn, 2024-06-05:
# packages in environment at C:\Users\User\anaconda3\envs\phy2:
#
# Name                    Version                   Build  Channel
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
aws-c-auth                0.7.22               ha390a07_2    conda-forge
aws-c-cal                 0.6.14               h750c3ff_1    conda-forge
aws-c-common              0.9.19               h2466b09_0    conda-forge
aws-c-compression         0.2.18               h750c3ff_6    conda-forge
aws-c-event-stream        0.4.2               he05106e_12    conda-forge
aws-c-http                0.8.1               h55c4ea9_17    conda-forge
aws-c-io                  0.14.8               ha255af7_5    conda-forge
aws-c-mqtt                0.10.4               heb8cf2b_4    conda-forge
aws-c-s3                  0.5.9                haf5e0f7_3    conda-forge
aws-c-sdkutils            0.1.16               h750c3ff_2    conda-forge
aws-checksums             0.1.18               h750c3ff_6    conda-forge
aws-crt-cpp               0.26.9               hb2f7e64_0    conda-forge
aws-sdk-cpp               1.11.329             h752cf48_3    conda-forge
bokeh                     3.4.1              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                hcfcfb64_1    conda-forge
brotli-bin                1.1.0                hcfcfb64_1    conda-forge
brotli-python             1.1.0           py312h53d5487_1    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-ares                    1.28.1               hcfcfb64_0    conda-forge
ca-certificates           2024.6.2             h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
click                     8.1.7           win_pyh7428d3b_0    conda-forge
cloudpickle               3.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
colorcet                  3.1.0                    pypi_0    pypi
comm                      0.2.2              pyhd8ed1ab_0    conda-forge
contourpy                 1.2.1           py312h0d7def4_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
cython                    3.0.10          py312h53d5487_0    conda-forge
cytoolz                   0.12.3          py312he70551f_0    conda-forge
dask                      2024.5.2           pyhd8ed1ab_0    conda-forge
dask-core                 2024.5.2           pyhd8ed1ab_0    conda-forge
dask-expr                 1.1.2              pyhd8ed1ab_0    conda-forge
debugpy                   1.8.1           py312h53d5487_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
distributed               2024.5.2           pyhd8ed1ab_0    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
fonttools                 4.53.0          py312h4389bb4_0    conda-forge
freetype                  2.12.1               hdaf720e_2    conda-forge
freetype-py               2.4.0                    pypi_0    pypi
fsspec                    2024.5.0           pyhff2d567_0    conda-forge
gettext                   0.22.5               h5728263_2    conda-forge
gettext-tools             0.22.5               h7d00a51_2    conda-forge
ghp-import                2.1.0                    pypi_0    pypi
git                       2.45.1               h57928b3_0    conda-forge
glib                      2.80.2               h0df6a38_0    conda-forge
glib-tools                2.80.2               h2f9d560_0    conda-forge
gst-plugins-base          1.22.9               h001b923_1    conda-forge
gstreamer                 1.22.9               hb4038d2_1    conda-forge
h5py                      3.11.0          nompi_py312ha036244_101    conda-forge
hdf5                      1.14.3          nompi_h2b43c12_103    conda-forge
hsluv                     5.0.4                    pypi_0    pypi
icu                       73.2                 h63175ca_0    conda-forge
idna                      3.7                pyhd8ed1ab_0    conda-forge
importlib-metadata        7.1.0              pyha770c72_0    conda-forge
importlib_metadata        7.1.0                hd8ed1ab_0    conda-forge
iniconfig                 2.0.0              pyhd8ed1ab_0    conda-forge
intel-openmp              2024.1.0           h57928b3_966    conda-forge
ipykernel                 6.29.3             pyha63f2e9_0    conda-forge
ipython                   8.25.0             pyh7428d3b_0    conda-forge
jedi                      0.19.1             pyhd8ed1ab_0    conda-forge
jinja2                    3.1.4              pyhd8ed1ab_0    conda-forge
[... 144 lines cut ...]
tornado                   6.4             py312he70551f_0    conda-forge
tqdm                      4.66.4                   pypi_0    pypi
traitlets                 5.14.3             pyhd8ed1ab_0    conda-forge
types-pyyaml              6.0.12.20240311    pyhd8ed1ab_0    conda-forge
typing_extensions         4.12.1             pyha770c72_0    conda-forge
tzdata                    2024a                h0c530f3_0    conda-forge
[... cut ...]
xyzservices               2024.4.0           pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.5                he1f189c_4    conda-forge
zict                      3.0.0              pyhd8ed1ab_0    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge

zm711, 2024-06-05:
Does openGL work on your Intel x-series? Seems like the error is happening there.

AbhiSwamiUConn, 2024-06-05:
How do I check that openGL works correctly?

zm711, 2024-06-05:
I would just google. For example Apple deprecated the use of openGL on macOS years ago which makes it very hard/basically impossible at this point to run phy on an M-series Mac. So I would google your exact processor and see if there are any issues with openGL on your processor/ in Windows 10. 

I've got Phy working on a Windows ten, but on the standard i-series processors, so I just wasn't sure if the x-series were the problem.

AbhiSwamiUConn, 2024-06-05:
From what I see, intel x-series processors themselves do not have graphics support.  However, the computer I am using does appear to have a graphics card that supports at least openGL 4.5.  Second, when I look in windows settings, hardware accelerated gpu scheduling is off.

zm711, 2024-06-05:
```
name: phy2_test
channels:
  - conda-forge
  - defaults
dependencies:
  - python=3.11
  - pip
  - git
  - numpy
  - matplotlib
  - scipy
  - h5py
  - pyqt
  - pyopengl
  - pyqtwebengine
  - pytest
  - qtconsole
  - requests
  - responses
  - traitlets
  - dask
  - cython
  - pillow
  - scikit-learn
  - joblib
  - pip:
    - git+https://github.com/cortex-lab/phy.git
```

could you try using this as your yaml instead. We recently tried adding 3.12 support, but I think it might not be fully there yet.

zm711, 2024-06-06:
Thank you! It helped me discover that there is still an issue with 3.12. I changed the yaml file to automatically just do python 3.11 until I can track down the issue.

jesusdpa1, 2025-06-12:
[image]

For anyone looking into this, at least on macOS M1 the issue seems to be with opengl

when setting up the startup of phy like this it works, now I am still unable to use manual clustering, 

```py
import os
from pathlib import Path

from PyQt5.QtGui import QSurfaceFormat

from phy.apps.template import template_gui

# Set environment variables BEFORE importing QtWebEngine
os.environ['QTWEBENGINE_CHROMIUM_FLAGS'] = (
    '--enable-gpu --use-gl=desktop --enable-features=VaapiVideoDecoder'
)
os.environ['QTWEBENGINE_DISABLE_SANDBOX'] = '1'

# Configure OpenGL for both PyQt5 and QtWebEngine
def setup_opengl_for_webengine():
    # This must be called before QApplication creation
    # QApplication.setAttribute(Qt.AA_ShareOpenGLContexts, True)

    format = QSurfaceFormat()
    format.setVersion(2, 1)  # Mac M1 maximum
    format.setProfile(QSurfaceFormat.CoreProfile)
    format.setDepthBufferSize(24)
    format.setStencilBufferSize(8)
    format.setSamples(4)
    format.setSwapBehavior(QSurfaceFormat.DoubleBuffer)

    QSurfaceFormat.setDefaultFormat(format)

# Apply configuration
setup_opengl_for_webengine()
# Now all OpenGL contexts created will use this format
# %%
HOME_PATH = Path().home()
BASE_PATH = HOME_PATH / 'Documents'
PARAMS_PATH = BASE_PATH / 'ks4_output_2025-06-10_19-27' / 'sorter_output' / 'params.py'

# %%
template_gui(PARAMS_PATH)

```

```
An OpenGL surfcace format was requested that is either not version 3.2 or higher or a not Core Profile.
Chromium on macOS will fall back to software rendering in this case.
Hardware acceleration and features such as WebGL will not be available.
Sandboxing disabled by user.
```