cortex-lab/phy #1285: Extract waveform

Lorena-J, 2024-06-19:
Hello!!! I'm triyng to extract waveform information in my analysis and I'm using this code from the doc:
```python
model = load_model(params) 
waveforms = model.get_cluster_spike_waveforms(cluster_id)
n_spikes, n_samples, n_channels_loc = waveforms.shape
```

But I'm not sure about the shape, I understand what n_spikes is but the n_samples and n_channels_loc I don't quite understand what they are. 

There is another file (spike_times) that I understand that has the time in which a spike occurs, so if I take cluster 1 and all its spike_times I would have some way to relate that n_spikes of the variable 'waveforms' belongs to each spike time...?

zm711, 2024-06-20:
n_samples are time. Recording equipment doesn't use seconds (typically) instead it uses a counter of rising ints. The conversion between time and sample is `time = sample * sampling_rate`.. Sometimes samples are also called frames (like you would imagine for a movie). n_channel_loc would be the channels that the spike is on.

so the idea of the matrix is:

you have a series of spikes and each spike is defined as electrical activity over time (n_samples) and over space (you need a waveform for each channel that you're interested in).

Does that make sense?

Lorena-J, 2024-07-01:
Yes, it make sense! I need to create a .txt file with the milivolts points of each spike in time_spike files. I need to create a .txt file with the data in millivolts that make up each spike detected in the spike_times file. That is, if the first spike detected is at time 1, I need to know the mv that make up that spike. I'm not quite sure what files I need to get this information.

zm711, 2024-07-02:
Your code above looks to do it. This really depends on what you want to do. Again I have a preference for doing this specific type of work in spikeinterface because it gives you more filtering options and scaling options. In this case you read the data, but you're not necessarily in volts (mV or uV) so you can get the shape but maybe not the correct units.

Lorena-J, 2024-07-03:
Thank you very much for your help! now I have it all clear! :)