cortex-lab/phy #1286: Stuck on Adding default color schemes to TraceView.

1HosseinHeydari8, 2024-06-20:
Hello,
I have created the environment using "environment.yml".
The data I'm trying to visualize is from Kilosort/docs/tutorials/basic_example.ipynb (ZFM-02370_mini.imec0.ap.bin).
I fed that data to the kilosort GUI, and got some results:
[image]
Running the phy template-gui params.py gives me an error:
[image]
Downgrading numpy didn't help. So instead, I changed that line of code as suggested in the error. Running the gui again, this time I get this:
[image]
It has been stuck on the last line.

I appreciate any help.

zm711, 2024-06-20:
Downgrading to which numpy and which python version?

1HosseinHeydari8, 2024-06-20:
Thank you for the prompt response.
The Python version is 3.11.9.
I've tried numpy 1.25.0 and 1.26.0. I get this error in both cases:
[image]
However, when I check the installation, it is actually there (using pip list):
Package            Version
------------------ ---------------
asttokens          2.4.1
bokeh              3.4.1
Brotli             1.1.0
cached-property    1.5.2
certifi            2024.6.2
charset-normalizer 3.3.2
click              8.1.7
cloudpickle        3.0.0
colorama           0.4.6
colorcet           3.1.0
comm               0.2.2
contourpy          1.2.1
cycler             0.12.1
Cython             3.0.10
cytoolz            0.12.3
dask               2024.6.0
dask-expr          1.1.3
debugpy            1.8.1
decorator          5.1.1
distributed        2024.6.0
exceptiongroup     1.2.0
executing          2.0.1
fonttools          4.53.0
fsspec             2024.6.0
ghp-import         2.1.0
h5py               3.11.0
idna               3.7
importlib_metadata 7.1.0
iniconfig          2.0.0
ipykernel          6.29.4
ipython            8.25.0
jedi               0.19.1
Jinja2             3.1.4
joblib             1.4.2
jupyter_client     8.6.2
jupyter_core       5.7.2
kiwisolver         1.4.5
locket             1.0.0
lz4                4.3.3
Markdown           3.6
MarkupSafe         2.1.5
matplotlib         3.8.4
matplotlib-inline  0.1.7
mergedeep          1.3.4
mkdocs             1.6.0
mkdocs-get-deps    0.2.0
msgpack            1.0.8
mtscomp            1.0.2
munkres            1.1.4
nest_asyncio       1.6.0
numpy              1.25.0
packaging          24.1
pandas             2.2.2
parso              0.8.4
partd              1.4.2
pathspec           0.12.1
phy                2.0b6
phylib             2.6.0
pickleshare        0.7.5
pillow             10.3.0
pip                24.0
platformdirs       4.2.2
pluggy             1.5.0
ply                3.11
prompt_toolkit     3.0.47
psutil             5.9.8
pure-eval          0.2.2
pyarrow            16.1.0
pyarrow-hotfix     0.6
Pygments           2.18.0
PyOpenGL           3.1.6
pyparsing          3.1.2
PyQt5              5.15.9
PyQt5-sip          12.12.2
PyQtWebEngine      5.15.4
PySocks            1.7.1
pytest             8.2.2
python-dateutil    2.9.0
pytz               2024.1
pywin32            306
PyYAML             6.0.1
pyyaml_env_tag     0.1
pyzmq              26.0.3
qtconsole          5.5.2
QtPy               2.4.1
requests           2.32.3
responses          0.25.3
scikit-learn       1.5.0
scipy              1.13.1
setuptools         70.1.0
sip                6.7.12
six                1.16.0
sortedcontainers   2.4.0
stack-data         0.6.2
tblib              3.0.0
threadpoolctl      3.5.0
toml               0.10.2
tomli              2.0.1
toolz              0.12.1
tornado            6.4.1
tqdm               4.66.4
traitlets          5.14.3
types-PyYAML       6.0.12.20240311
typing_extensions  4.12.2
tzdata             2024.1
urllib3            2.2.2
watchdog           4.0.1
wcwidth            0.2.13
wheel              0.43.0
win-inet-pton      1.1.0
xyzservices        2024.6.0
zict               3.0.0
zipp               3.19.2

zm711, 2024-06-20:
Could you try reinstalling with a numpy limit in the yaml file. I wonder if the pickle and numpy are out of sync because of just downgrading rather than installing at the beginning. Sometimes pip-conda messes these things up. 

To be clear: in the file do

```yaml
name: phy2
channels:
  - conda-forge
  - defaults
dependencies:
  - python=3.11
  - pip
  - git
  - numpy<2.0
  - matplotlib
  - scipy
  - h5py
  - pyqt
  - pyopengl
  - pyqtwebengine
  - pytest
  - qtconsole
  - requests
  - responses
  - traitlets
  - dask
  - cython
  - pillow
  - scikit-learn
  - joblib
  - pip:
    - git+https://github.com/cortex-lab/phy.git
```

1HosseinHeydari8, 2024-06-20:
Thank you! It is working now:
[image]

zm711, 2024-06-20:
wonderful. The risks of pip + conda. haha. Have a nice day!

closed 2024-06-20 by zm711