cortex-lab/phy #1287: Segmentation fault when attempting to open Kilosort4 output in Phy

a-savoy, 2024-06-20:
I have no issues sorting and preparing the output to Phy using Kilosort4 (all default settings), via the Docker container with SpikeInterface or without using SpikeInterface. I created a Phy environment from the environment.yml file (the NumPy version was an issue, but it worked after I changed the version to 1.24.1). When I run `phy trace-gui params.py` there is no issue, but when I run `phy template-gui params.py`, I see the following, regardless of the dataset:

> 16:33:44.806 [W] model:667 Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> Segmentation fault (core dumped)

If I run Kilosort4 outside of SpikeInterface, sometimes Phy can open the output, but sometimes I get the same segmentation fault. 

Here is an example debug output:

> $ phy template-gui params.py --debug
> 14:14:35.547 [D] init:68 Start capturing exceptions.
> 14:14:35.731 [D] model:619 Loading spike clusters.
> 14:14:35.861 [D] model:569 No channel shank file found.
> 14:14:35.861 [D] model:692 Loading templates.
> 14:14:35.864 [D] model:720 Templates are sparse.
> 14:14:35.869 [W] model:667 Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 14:14:35.870 [D] model:730 Loading the whitening matrix.
> 14:14:35.870 [D] model:434 Whitening matrix file not found.
> 14:14:35.870 [D] model:737 Loading the inverse of the whitening matrix.
> 14:14:35.872 [D] model:766 Loading features.
> 14:14:35.873 [D] model:781 Features are sparse.
> 14:14:35.873 [D] model:803 Loading template features.
> 14:14:35.874 [D] model:504 Load cluster_group.tsv.
> 14:14:35.875 [D] model:504 Load cluster_si_unit_ids.tsv.
> 14:14:35.876 [D] model:504 Load cluster_channel_group.tsv.
> 14:14:36.174 [D] context:100 Initialize joblib cache dir at /phy/.phy.
> 14:14:36.174 [D] context:101 Reducing the size of the cache if needed.
> 14:14:36.176 [D] base:102 Add filter high_pass.
> 14:14:36.177 [D] config:31 Load config file /.phy/phy_config.py.
> 14:14:36.177 [D] plugin:146 Loading 0 plugins.
> 14:14:36.180 [D] context:209 The file /phy/.phy/new_cluster_id.pkl doesn't exist.
> 14:14:36.225 [D] context:185 Save data to /phy/.phy/spikes_per_cluster.pkl.
> 14:14:36.309 [D] gui:463 Creating GUI.
> 14:14:36.320 [D] state:46 Load /.phy/TemplateGUI/state.json for GUIState.
> Segmentation fault (core dumped)

Here are the packages in the the phy environment:

> _libgcc_mutex             0.1                        main  
> _openmp_mutex             5.1                       1_gnu  
> abseil-cpp                20211102.0           hd4dd3e8_0  
> arrow-cpp                 14.0.2               h374c478_1  
> asttokens                 2.0.5              pyhd3eb1b0_0  
> aws-c-auth                0.6.19               h5eee18b_0  
> aws-c-cal                 0.5.20               hdbd6064_0  
> aws-c-common              0.8.5                h5eee18b_0  
> aws-c-compression         0.2.16               h5eee18b_0  
> aws-c-event-stream        0.2.15               h6a678d5_0  
> aws-c-http                0.6.25               h5eee18b_0  
> aws-c-io                  0.13.10              h5eee18b_0  
> aws-c-mqtt                0.7.13               h5eee18b_0  
> aws-c-s3                  0.1.51               hdbd6064_0  
> aws-c-sdkutils            0.1.6                h5eee18b_0  
> aws-checksums             0.1.13               h5eee18b_0  
> aws-crt-cpp               0.18.16              h6a678d5_0  
> aws-sdk-cpp               1.10.55              h721c034_0  
> blas                      1.0                         mkl  
> bokeh                     3.4.1           py311h92b7b1e_0  
> boost-cpp                 1.82.0               hdb19cb5_2  
> bottleneck                1.3.7           py311hf4808d0_0  
> brotli                    1.0.9                h5eee18b_8  
> brotli-bin                1.0.9                h5eee18b_8  
> brotli-python             1.0.9           py311h6a678d5_8  
> bzip2                     1.0.8                h5eee18b_6  
> c-ares                    1.19.1               h5eee18b_0  
> ca-certificates           2024.3.11            h06a4308_0  
> certifi                   2024.6.2        py311h06a4308_0  
> charset-normalizer        2.0.4              pyhd3eb1b0_0  
> click                     8.1.7           py311h06a4308_0  
> cloudpickle               2.2.1           py311h06a4308_0  
[... 164 lines cut ...]
> stack_data                0.2.0              pyhd3eb1b0_0  
> tbb                       2021.8.0             hdb19cb5_0  
> tblib                     1.7.0              pyhd3eb1b0_0  
> threadpoolctl             2.2.0              pyh0d69192_0  
> tk                        8.6.14               h39e8969_0  
> toolz                     0.12.0          py311h06a4308_0  
> tornado                   6.3.3           py311h5eee18b_0  
> tqdm                      4.66.4                   pypi_0    pypi
> traitlets                 5.14.3          py311h06a4308_0  
> typing_extensions         4.11.0          py311h06a4308_0  
> tzdata                    2024a                h04d1e81_0  
> unicodedata2              15.1.0          py311h5eee18b_0  
> urllib3                   2.2.1           py311h06a4308_0  
> utf8proc                  2.6.1                h5eee18b_1  
> watchdog                  4.0.1                    pypi_0    pypi
> wcwidth                   0.2.5              pyhd3eb1b0_0  
> wheel                     0.43.0          py311h06a4308_0  
> xyzservices               2022.9.0        py311h06a4308_1  
> xz                        5.4.6                h5eee18b_1  
> yaml                      0.2.5                h7b6447c_0  
> zeromq                    4.3.5                h6a678d5_0  
> zict                      3.0.0           py311h06a4308_0  
> zipp                      3.17.0          py311h06a4308_0  
> zlib                      1.2.13               h5eee18b_1  
> zstd                      1.5.5                hc292b87_2

a-savoy, 2024-06-20:
Ubuntu 22.04.2 LTS; yaml level.

zm711, 2024-06-20:
Do you get any pop errors or any others in the terminal when this happens or does the seg fault happen silently?

a-savoy, 2024-06-20:
Silently.

zm711, 2024-06-20:
Could you post the contents of `phy.log` and `params.py`. Feel free to edit file paths for privacy but could you put ... so I know where you've edited. The phy.log should give us similar info: For example on a good loading of my data I see this:

```
90m12:59:31.982 [D] state:46             Load C:\Users\ZacharyMcKenzie\.phy\TemplateGUI\state.json for GUIState.[0m
[90m12:59:32.429 [D] gui:718              Add view ClusterView to GUI.[0m
```

As the line where you seg faulted.

a-savoy, 2024-06-20:
params.py
> dat_path = r'/.../phy/recording.dat'
> n_channels_dat = 20
> dtype = 'int16'
> offset = 0
> sample_rate = 30000.0
> hp_filtered = True

phy.log

> [90m20:01:34.602 [D] context:80           Create cache directory `.../phy/.phy`.[0m
> [90m20:01:34.602 [D] context:80           Create cache directory `.../phy/.phy`.[0m
> [90m20:01:35.166 [D] context:100          Initialize joblib cache dir at `.../phy/.phy`.[0m
> [90m20:01:35.166 [D] context:100          Initialize joblib cache dir at `.../phy/.phy`.[0m
> [90m20:01:35.166 [D] context:101          Reducing the size of the cache if needed.[0m
> [90m20:01:35.166 [D] context:101          Reducing the size of the cache if needed.[0m
> [90m20:01:35.198 [D] base:102             Add filter `high_pass`.[0m
> [90m20:01:35.198 [D] base:102             Add filter `high_pass`.[0m
> [90m20:01:35.199 [D] config:31            Load config file `/home/labadmin/.phy/phy_config.py`.[0m
> [90m20:01:35.199 [D] config:31            Load config file `/home/labadmin/.phy/phy_config.py`.[0m
> [90m20:01:35.211 [D] plugin:146           Loading 0 plugins.[0m
> [90m20:01:35.211 [D] plugin:146           Loading 0 plugins.[0m
> [90m20:01:35.217 [D] context:209          The file `.../phy/.phy/new_cluster_id.pkl` doesn't exist.[0m
> [90m20:01:35.217 [D] context:209          The file `.../phy/.phy/new_cluster_id.pkl` doesn't exist.[0m
> [90m20:01:35.218 [D] context:209          The file `.../phy/.phy/spikes_per_cluster.pkl` doesn't exist.[0m
> [90m20:01:35.218 [D] context:209          The file `.../phy/.phy/spikes_per_cluster.pkl` doesn't exist.[0m
> [90m20:01:35.258 [D] clustering:237       Recompute spikes_per_cluster manually: this might take a while.[0m
> [90m20:01:35.258 [D] clustering:237       Recompute spikes_per_cluster manually: this might take a while.[0m
> [90m20:01:35.422 [D] context:185          Save data to `.../phy/.phy/spikes_per_cluster.pkl`.[0m
> [90m20:01:35.422 [D] context:185          Save data to `.../phy/.phy/spikes_per_cluster.pkl`.[0m
> [90m20:01:35.495 [D] gui:463              Creating GUI.[0m
> [90m20:01:35.495 [D] gui:463              Creating GUI.[0m
> [90m20:01:35.770 [D] state:46             Load /home/labadmin/.phy/TemplateGUI/state.json for GUIState.[0m
> [90m20:01:35.770 [D] state:46             Load /home/labadmin/.phy/TemplateGUI/state.json for GUIState.[0m
> [90m14:13:43.512 [D] context:100          Initialize joblib cache dir at `.../phy/.phy`.[0m
> [90m14:13:43.512 [D] context:100          Initialize joblib cache dir at `.../phy/.phy`.[0m
> [90m14:13:43.513 [D] context:101          Reducing the size of the cache if needed.[0m
> [90m14:13:43.513 [D] context:101          Reducing the size of the cache if needed.[0m
> [90m14:13:43.517 [D] base:102             Add filter `high_pass`.[0m
> [90m14:13:43.517 [D] base:102             Add filter `high_pass`.[0m
> [90m14:13:43.518 [D] config:31            Load config file `/home/labadmin/.phy/phy_config.py`.[0m
> [90m14:13:43.518 [D] config:31            Load config file `/home/labadmin/.phy/phy_config.py`.[0m
> [90m14:13:43.518 [D] plugin:146           Loading 0 plugins.[0m
> [90m14:13:43.518 [D] plugin:146           Loading 0 plugins.[0m
> [90m14:13:43.521 [D] context:209          The file `.../phy/.phy/new_cluster_id.pkl` doesn't exist.[0m
> [90m14:13:43.521 [D] context:209          The file `.../phy/.phy/new_cluster_id.pkl` doesn't exist.[0m
> [90m14:13:43.620 [D] context:185          Save data to `.../phy/.phy/spikes_per_cluster.pkl`.[0m
> [90m14:13:43.620 [D] context:185          Save data to `.../phy/.phy/spikes_per_cluster.pkl`.[0m
> [90m14:13:43.730 [D] gui:463              Creating GUI.[0m
> [90m14:13:43.730 [D] gui:463              Creating GUI.[0m
> [90m14:13:43.744 [D] state:46             Load /home/labadmin/.phy/TemplateGUI/state.json for GUIState.[0m
> [90m14:13:43.744 [D] state:46             Load /home/labadmin/.phy/TemplateGUI/state.json for GUIState.[0m
> [90m14:14:36.174 [D] context:100          Initialize joblib cache dir at `.../phy/.phy`.[0m
> [90m14:14:36.174 [D] context:100          Initialize joblib cache dir at `.../phy/.phy`.[0m
> [90m14:14:36.174 [D] context:101          Reducing the size of the cache if needed.[0m
> [90m14:14:36.174 [D] context:101          Reducing the size of the cache if needed.[0m
> [90m14:14:36.176 [D] base:102             Add filter `high_pass`.[0m
> [90m14:14:36.176 [D] base:102             Add filter `high_pass`.[0m
> [90m14:14:36.177 [D] config:31            Load config file `/home/labadmin/.phy/phy_config.py`.[0m
> [90m14:14:36.177 [D] config:31            Load config file `/home/labadmin/.phy/phy_config.py`.[0m
> [90m14:14:36.177 [D] plugin:146           Loading 0 plugins.[0m
> [90m14:14:36.177 [D] plugin:146           Loading 0 plugins.[0m
> [90m14:14:36.180 [D] context:209          The file `.../phy/.phy/new_cluster_id.pkl` doesn't exist.[0m
> [90m14:14:36.180 [D] context:209          The file `.../phy/.phy/new_cluster_id.pkl` doesn't exist.[0m
> [90m14:14:36.225 [D] context:185          Save data to `.../phy/.phy/spikes_per_cluster.pkl`.[0m
> [90m14:14:36.225 [D] context:185          Save data to `.../phy/.phy/spikes_per_cluster.pkl`.[0m
> [90m14:14:36.309 [D] gui:463              Creating GUI.[0m
> [90m14:14:36.309 [D] gui:463              Creating GUI.[0m
> [90m14:14:36.320 [D] state:46             Load /home/labadmin/.phy/TemplateGUI/state.json for GUIState.[0m
> [90m14:14:36.320 [D] state:46             Load /home/labadmin/.phy/TemplateGUI/state.json for GUIState.[0m
>

zm711, 2024-06-20:
Also I want to confirm what you mean at the beginning here:

> I have no issues sorting and preparing the output to Phy using Kilosort4 (all default settings), via the Docker container with SpikeInterface or without using SpikeInterface.

Do you mean that all formats fail with Phy (no matter how you run KS4) or that one of them works? (or to ask another way is phy globally failing or failing for specific use cases).

a-savoy, 2024-06-20:
Sorry, I mean that all outputs from SpikeInterface fail. Most outputs run without issue if they are from Kilosort4 on its own (not via SpikeInterface), but some also fail with a segmentation fault. However, Phy loads any output in the trace-gui, so all of these issues apply only to the template-gui.

zm711, 2024-06-20:
It's weird because the issue is occurring while we are still making the gui. Sometimes we get index errors with kilosort because it extra spikes (past recording edge) sometimes. Are you doing `export_to_phy` or just using the wrapper from SI? `export_to_phy` has us write the files whereas the wrapper should just use native KS stuff. 

I'm trying to track this down to:
KS problem
SpikeInterface export problem
SpikeInterface wrapper problem
Phy problem

I had thought it was Phy. The best way to test would be could you try spikesorting the same data with mountainsort5, spykingcircus 2 or TDC2-then export to phy and then try to load with phy and see what happens?

If we can't piece this together we may need you to share a "fail" dataset that failed even from native KS4 to try to track this down. What is the binary file you're using when you use native KS4? Could you copy the params.py from a native KS4 file that failed and one that worked?

a-savoy, 2024-06-20:
If it helps, here are the output files when I use Kilosort4 alone, not with SpikeInterface:

amplitudes.npy
channel_map.npy
channel_positions.npy
cluster_Amplitude.tsv
cluster_ContamPct.tsv
cluster_group.tsv
cluster_KSLabel.tsv
kept_spikes.npy
ops.npy
params.py
pc_feature_ind.npy
pc_features.npy
phy.log
similar_templates.npy
spike_clusters.npy
spike_detection_templates.npy
spike_positions.npy
spike_templates.npy
spike_times.npy
templates.npy
templates_ind.npy
whitening_mat.npy
whitening_mat_dat.npy
whitening_mat_inv.npy

zm711, 2024-06-20:
So could you try sorting with a different sorter, do export to phy and then see if it repeats. Then maybe we narrow back down to spikeinterface and we actually switch back to your other issue. If it is fine with the other sorter than it might be how spikeinterface is exporting the KS4 data itself (still an SI issue, but KS4 has been changing so much it is tricky to keep up).

a-savoy, 2024-06-20:
**Update:** The KS4 output files that I was previously able to load in Phy are now giving me the segmentation fault. Maybe this is an issue with my system? Could it be memory related?

zm711, 2024-06-21:
Typically seg faults occur with things working at the c-api level. This could be something between python-numpy-qt (but looking at your python and numpy I think you should be fine there). The other issue could be that KS is returning some sort of value that is leading the gui to try to access memory it shouldn't. 

so I have to go back to my previous statement. We need to test another sorter and exporting to phy.

a-savoy, 2024-06-22:
I tested SpyKING CIRCUS, but I get the same error:

> 19:13:51.642 [D] __init__:68          Start capturing exceptions.
> 19:13:51.780 [D] model:619            Loading spike clusters.
> 19:13:51.877 [D] model:569            No channel shank file found.
> 19:13:51.877 [D] model:692            Loading templates.
> 19:13:51.879 [D] model:720            Templates are sparse.
> 19:13:51.884 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 19:13:51.884 [D] model:730            Loading the whitening matrix.
> 19:13:51.884 [D] model:434            Whitening matrix file not found.
> 19:13:51.884 [D] model:737            Loading the inverse of the whitening matrix.
> 19:13:51.886 [D] model:766            Loading features.
> 19:13:51.887 [D] model:781            Features are sparse.
> 19:13:51.887 [D] model:803            Loading template features.
> 19:13:51.888 [D] model:504            Load `cluster_group.tsv`.
> 19:13:51.889 [D] model:504            Load `cluster_si_unit_ids.tsv`.
> 19:13:51.889 [D] model:504            Load `cluster_channel_group.tsv`.
> 19:13:52.166 [D] context:100          Initialize joblib cache dir at `/.../phy_SC/.phy`.
> 19:13:52.166 [D] context:101          Reducing the size of the cache if needed.
> 19:13:52.168 [D] base:102             Add filter `high_pass`.
> 19:13:52.169 [D] config:31            Load config file `/.../.phy/phy_config.py`.
> 19:13:52.169 [D] plugin:146           Loading 0 plugins.
> 19:13:52.171 [D] context:209          The file `/.../phy_SC/.phy/new_cluster_id.pkl` doesn't exist.
> 19:13:52.217 [D] context:185          Save data to `/...phy_SC/.phy/spikes_per_cluster.pkl`.
> 19:13:52.295 [D] gui:463              Creating GUI.
> 19:13:52.304 [D] state:46             Load /.../.phy/TemplateGUI/state.json for GUIState.
> Segmentation fault (core dumped)

410pfeliciano, 2024-06-22:
I'm having a similar problem when opening phy2 on Ubuntu 22.04 but not in Win10. 

**19:07:23.349 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
Segmentation fault (core dumped)**

Here is my Conda env list: # Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
abseil-cpp                20211102.0           hd4dd3e8_0  
arrow-cpp                 14.0.2               h374c478_1  
asttokens                 2.0.5              pyhd3eb1b0_0  
aws-c-auth                0.6.19               h5eee18b_0  
aws-c-cal                 0.5.20               hdbd6064_0  
aws-c-common              0.8.5                h5eee18b_0  
aws-c-compression         0.2.16               h5eee18b_0  
aws-c-event-stream        0.2.15               h6a678d5_0  
aws-c-http                0.6.25               h5eee18b_0  
aws-c-io                  0.13.10              h5eee18b_0  
aws-c-mqtt                0.7.13               h5eee18b_0  
aws-c-s3                  0.1.51               hdbd6064_0  
aws-c-sdkutils            0.1.6                h5eee18b_0  
aws-checksums             0.1.13               h5eee18b_0  
aws-crt-cpp               0.18.16              h6a678d5_0  
aws-sdk-cpp               1.10.55              h721c034_0  
blas                      1.0                         mkl  
bokeh                     3.4.1           py311h92b7b1e_0  
boost-cpp                 1.82.0               hdb19cb5_2  
bottleneck                1.3.7           py311hf4808d0_0  
brotli                    1.0.9                h5eee18b_8  
brotli-bin                1.0.9                h5eee18b_8  
brotli-python             1.0.9           py311h6a678d5_8  
bzip2                     1.0.8                h5eee18b_6  
c-ares                    1.19.1               h5eee18b_0  
ca-certificates           2024.3.11            h06a4308_0  
certifi                   2024.6.2        py311h06a4308_0  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
click                     8.1.7           py311h06a4308_0  
cloudpickle               2.2.1           py311h06a4308_0  
colorcet                  3.1.0                    pypi_0    pypi
comm                      0.2.1           py311h06a4308_0  
contourpy                 1.2.0           py311hdb19cb5_0  
cycler                    0.11.0             pyhd3eb1b0_0  
cyrus-sasl                2.1.28               h52b45da_1  
cython                    3.0.10          py311h5eee18b_0  
cytoolz                   0.12.2          py311h5eee18b_0  
dask                      2024.5.0        py311h06a4308_0  
dask-core                 2024.5.0        py311h06a4308_0  
dask-expr                 1.1.0           py311h06a4308_0  
dbus                      1.13.18              hb2f20db_0  
debugpy                   1.6.7           py311h6a678d5_0  
decorator                 5.1.1              pyhd3eb1b0_0  
distributed               2024.5.0        py311h06a4308_0  
executing                 0.8.3              pyhd3eb1b0_0  
expat                     2.6.2                h6a678d5_0  
fontconfig                2.14.1               h4c34cd2_2  
fonttools                 4.51.0          py311h5eee18b_0  
freetype                  2.12.1               h4a9f257_0  
fsspec                    2024.3.1        py311h06a4308_0  
gflags                    2.2.2                h6a678d5_1  
ghp-import                2.1.0                    pypi_0    pypi
glib                      2.78.4               h6a678d5_0  
glib-tools                2.78.4               h6a678d5_0  
glog                      0.5.0                h6a678d5_1  
grpc-cpp                  1.48.2               he1ff14a_1  
gst-plugins-base          1.14.1               h6a678d5_1  
gstreamer                 1.14.1               h5eee18b_1  
h5py                      3.11.0          py311h865a13c_0  
hdf5                      1.12.1               h2b7332f_3  
heapdict                  1.0.1              pyhd3eb1b0_0  
icu                       73.1                 h6a678d5_0  
[... 133 lines cut ...]
stack_data                0.2.0              pyhd3eb1b0_0  
tbb                       2021.8.0             hdb19cb5_0  
tblib                     1.7.0              pyhd3eb1b0_0  
threadpoolctl             2.2.0              pyh0d69192_0  
tk                        8.6.14               h39e8969_0  
toolz                     0.12.0          py311h06a4308_0  
tornado                   6.4.1           py311h5eee18b_0  
tqdm                      4.66.4                   pypi_0    pypi
traitlets                 5.14.3          py311h06a4308_0  
typing_extensions         4.11.0          py311h06a4308_0  
tzdata                    2024a                h04d1e81_0  
unicodedata2              15.1.0          py311h5eee18b_0  
urllib3                   2.2.2           py311h06a4308_0  
utf8proc                  2.6.1                h5eee18b_1  
watchdog                  4.0.1                    pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0  
wheel                     0.43.0          py311h06a4308_0  
xyzservices               2022.9.0        py311h06a4308_1  
xz                        5.4.6                h5eee18b_1  
yaml                      0.2.5                h7b6447c_0  
zeromq                    4.3.5                h6a678d5_0  
zict                      3.0.0           py311h06a4308_0  
zipp                      3.17.0          py311h06a4308_0  
zlib                      1.2.13               h5eee18b_1  
zstd                      1.5.5                hc292b87_2

zm711, 2024-06-23:
I think this is either a python point release issue. Last time I checked it was 3.11.4 or .5 I think and we are onto 3.11.9. Could you try installing with python 3.10 instead and see if that fixes it. 

The other potential issue would be a point release in any of the qt packages. Until I get the CI working I don't have access to a linux system to test this myself. So If you're willing to put in the time to find the exact point releases we need for linux we can limit this in the install instruction. But since this works on Windows I'm not sure how I can recreate the python + qt stuff to find the exact problem.

410pfeliciano, 2024-06-23:
I installed python 3.10 but that did not solved the problem. Here is the debug info:

13:38:11.975 [D] __init__:68          Start capturing exceptions.
13:38:12.051 [D] model:619            Loading spike clusters.
13:38:12.115 [D] model:569            No channel shank file found.
13:38:12.116 [D] model:692            Loading templates.
13:38:12.116 [D] model:724            Templates are dense.
13:38:12.117 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
13:38:12.118 [D] model:730            Loading the whitening matrix.
13:38:12.118 [D] model:737            Loading the inverse of the whitening matrix.
13:38:12.118 [D] model:766            Loading features.
13:38:12.119 [D] model:781            Features are sparse.
13:38:12.119 [D] model:803            Loading template features.
13:38:12.123 [D] model:530            Load spike_positions.npy.
13:38:12.124 [D] model:530            Load spike_detection_templates.npy.
13:38:12.124 [D] model:504            Load `cluster_ContamPct.tsv`.
13:38:12.124 [D] model:504            Load `cluster_Amplitude.tsv`.
13:38:12.125 [D] model:504            Load `cluster_KSLabel.tsv`.
13:38:12.125 [D] model:504            Load `cluster_group.tsv`.
13:38:12.410 [D] context:100          Initialize joblib cache dir at `/media/Pract/kilo_4_result/.phy`.
13:38:12.410 [D] context:101          Reducing the size of the cache if needed.
13:38:12.429 [D] base:102             Add filter `high_pass`.
13:38:12.429 [D] config:31            Load config file `/home/mouse3/.phy/phy_config.py`.
13:38:12.429 [D] plugin:146           Loading 0 plugins.
13:38:12.429 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
13:38:12.430 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
13:38:12.430 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
13:38:12.430 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
13:38:12.430 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
13:38:12.430 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
13:38:12.430 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
13:38:12.430 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
13:38:12.431 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
13:38:12.431 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
13:38:12.431 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
13:38:12.431 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
13:38:12.431 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
13:38:12.431 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
13:38:12.447 [D] context:185          Save data to `/media/mouse3/Samsung_T5/Pract/kilo_4_result/.phy/spikes_per_cluster.pkl`.
13:38:12.463 [D] gui:463              Creating GUI.
13:38:12.468 [D] state:46             Load /home/mouse3/.phy/TemplateGUI/state.json for GUIState.
13:38:12.555 [D] gui:718              Add view ClusterView to GUI.
13:38:12.561 [D] gui:718              Add view SimilarityView to GUI.
13:38:12.588 [D] gui:718              Add view WaveformView to GUI.
13:38:12.590 [D] base:337             Set state for WaveformView.
13:38:12.598 [D] gui:718              Add view CorrelogramView to GUI.
13:38:12.599 [D] base:337             Set state for CorrelogramView.
13:38:12.613 [D] gui:718              Add view ISIView to GUI.
13:38:12.614 [D] base:337             Set state for ISIView.
13:38:12.622 [D] gui:718              Add view FeatureView to GUI.
13:38:12.623 [D] base:337             Set state for FeatureView.
13:38:12.636 [D] gui:718              Add view AmplitudeView to GUI.
13:38:12.637 [D] base:337             Set state for AmplitudeView.
Segmentation fault (core dumped)

410pfeliciano, 2024-06-23:
The following installation(https://github.com/cortex-lab/phy/issues/1283#issuecomment-2150453627) solve my problem in Ubuntu 22.04:

name: phy2_test
channels:
        -conda-forge
        -defaults
dependencies:
        -python=3.11
        -pip
        -git
       -numpy=1.26.4
       -matplotlib
  -scipy
  -h5py
  -pyqt
  -pyopengl
  -pyqtwebengine
  -pytest
  -qtconsole
  -requests
  -responses
  -traitlets
  -dask
  -cython
  -pillow
  -scikit-learn
  -joblib
  -pip:
    -git+https://github.com/cortex-lab/phy.git

a-savoy, 2024-06-23:
**Update:** After restarting my computer (which runs Ubuntu 22.04, and Python 3.11 in my SI, KS, and Phy envs), I actually can load most outputs into Phy, regardless of whether or not I used SI. However, in most cases I get a new message upon loading the data (see below). Also, importantly, no waveforms are loaded and the columns for KSLabel, ContamPct, and Amplitude are missing. In a few cases, I get a new kind of error (see second readout below). I didn't update or change any environments or versions, so I'm not sure why restarting the computer resulted in such different behavior, or why I get varied behavior: out of nine datasets, one loads and shows waveforms; six load but don't show waveforms; and two don't load and give the error below (see table below). I have yet to get the segmentation fault again after restarting.

> $ phy template-gui params.py
> 19:54:28.448 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 0.01s - Debugger warning: It seems that frozen modules are being used, which may
> 0.00s - make the debugger miss breakpoints. Please pass -Xfrozen_modules=off
> 0.00s - to python to disable frozen modules.
> 0.00s - Note: Debugging will proceed. Set PYDEVD_DISABLE_FILE_VALIDATION=1 to disable this validation.

> $ phy template-gui params.py
> 19:52:56.501 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 19:52:56.712 [E] __init__:62          An error has occurred (ModuleNotFoundError): No module named 'numpy._core'
> Traceback (most recent call last):
>   File "/.../phy", line 8, in <module>
>     sys.exit(phycli())
>              ^^^^^^^^
>   File "/.../core.py", line 1157, in __call__
>     return self.main(*args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../core.py", line 1078, in main
>     rv = self.invoke(ctx)
>          ^^^^^^^^^^^^^^^^
>   File "/.../core.py", line 1688, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../core.py", line 1434, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../core.py", line 783, in invoke
>     return __callback(*args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../decorators.py", line 33, in new_func
>     return f(get_current_context(), *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../__init__.py", line 159, in cli_template_gui
>     template_gui(params_path, **kwargs)
>   File "/.../gui.py", line 217, in template_gui
>     controller = TemplateController(model=model, dir_path=dir_path, **kwargs)
>                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../base.py", line 470, in __init__
>     super(TemplateMixin, self).__init__(*args, **kwargs)
>   File "/.../base.py", line 926, in __init__
>     self._set_supervisor()
>   File "/.../gui.py", line 95, in _set_supervisor
>     super(TemplateController, self)._set_supervisor()
>   File "/.../base.py", line 1010, in _set_supervisor
>     supervisor = Supervisor(
>                  ^^^^^^^^^^^
>   File "/.../supervisor.py", line 637, in __init__
>     spc = context.load('spikes_per_cluster') if context else None
>           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../context.py", line 208, in load
>     return load_pickle(path)
>            ^^^^^^^^^^^^^^^^^
>   File "/.../_misc.py", line 144, in load_pickle
>     return load(path)
>            ^^^^^^^^^^
>   File "/.../numpy_pickle.py", line 658, in load
>     obj = _unpickle(fobj, filename, mmap_mode)
>           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../numpy_pickle.py", line 577, in _unpickle
>     obj = unpickler.load()
>           ^^^^^^^^^^^^^^^^
>   File "/.../pickle.py", line 1213, in load
>     dispatch[key[0]](self)
>   File "/.../pickle.py", line 1538, in load_stack_global
>     self.append(self.find_class(module, name))
>                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/.../pickle.py", line 1580, in find_class
>     __import__(module, level=0)
> ModuleNotFoundError: No module named 'numpy._core'

<google-sheets-html-origin>
Exp1 | Sorted by KS4 in SI | Sorted by KS4 (no SI) | SI Output Loads in Phy? | KS4-only Output Loads in Phy?
-- | -- | -- | -- | --
1 | No | Yes |   | Yes
2 | Yes | Yes |   | Yes, but no waveforms
3 | No | Yes |   | Yes, but no waveforms
4 | No | Yes |   | Yes, but no waveforms
5 | No | Yes |   | Yes, but no waveforms
6 | No | Yes |   | No, error 2
7 | Yes | No | Yes, but no waveforms |  
8 | Yes | No | Yes, but no waveforms |  
9 | Yes | No | No, error 2 |

a-savoy, 2024-06-23:
@410pfeliciano I created a new environment based on what you shared above, but it did not solve the problem for me. Here is the debug report:

> $ phy template-gui params.py
> 14:48:13.979 [D] __init__:68          Start capturing exceptions.
> 14:48:14.055 [D] model:619            Loading spike clusters.
> 14:48:14.107 [D] model:569            No channel shank file found.
> 14:48:14.108 [D] model:692            Loading templates.
> 14:48:14.110 [D] model:720            Templates are sparse.
> 14:48:14.112 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 14:48:14.112 [D] model:730            Loading the whitening matrix.
> 14:48:14.112 [D] model:434            Whitening matrix file not found.
> 14:48:14.113 [D] model:737            Loading the inverse of the whitening matrix.
> 14:48:14.114 [D] model:766            Loading features.
> 14:48:14.116 [D] model:781            Features are sparse.
> 14:48:14.116 [D] model:803            Loading template features.
> 14:48:14.117 [D] model:504            Load `cluster_group.tsv`.
> 14:48:14.118 [D] model:504            Load `cluster_si_unit_ids.tsv`.
> 14:48:14.118 [D] model:504            Load `cluster_channel_group.tsv`.
> 14:48:14.329 [D] context:100          Initialize joblib cache dir at `/.../.phy`.
> 14:48:14.329 [D] context:101          Reducing the size of the cache if needed.
> 14:48:14.332 [D] base:102             Add filter `high_pass`.
> 14:48:14.332 [D] config:31            Load config file `/.../phy_config.py`.
> 14:48:14.333 [D] plugin:146           Loading 0 plugins.
> 14:48:14.333 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
> 14:48:14.334 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
> 14:48:14.334 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
> 14:48:14.334 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
> 14:48:14.334 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
> 14:48:14.335 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
> 14:48:14.335 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
> 14:48:14.335 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
> 14:48:14.336 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
> 14:48:14.336 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
> 14:48:14.336 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
> 14:48:14.337 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
> 14:48:14.337 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
> 14:48:14.337 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
> 14:48:14.339 [D] context:209          The file `/.../new_cluster_id.pkl` doesn't exist.
> 14:44:10.755 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 14:44:11.491 [E] __init__:62          An error has occurred (ModuleNotFoundError): No module named 'numpy._core'
> Traceback (most recent call last):
>   File "/.../phy", line 8, in <module>
>     sys.exit(phycli())
>   File "/.../core.py", line 1157, in __call__
>     return self.main(*args, **kwargs)
>   File "/.../core.py", line 1078, in main
>     rv = self.invoke(ctx)
>   File "/.../core.py", line 1688, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/.../core.py", line 1434, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/.../core.py", line 783, in invoke
>     return __callback(*args, **kwargs)
>   File "/.../decorators.py", line 33, in new_func
>     return f(get_current_context(), *args, **kwargs)
>   File "/.../__init__.py", line 159, in cli_template_gui
>     template_gui(params_path, **kwargs)
>   File "/.../gui.py", line 217, in template_gui
>     controller = TemplateController(model=model, dir_path=dir_path, **kwargs)
>   File "/.../base.py", line 470, in __init__
>     super(TemplateMixin, self).__init__(*args, **kwargs)
>   File "/.../base.py", line 926, in __init__
>     self._set_supervisor()
>   File "/.../gui.py", line 95, in _set_supervisor
>     super(TemplateController, self)._set_supervisor()
>   File "/.../base.py", line 1010, in _set_supervisor
>     supervisor = Supervisor(
>   File "/.../supervisor.py", line 637, in __init__
>     spc = context.load('spikes_per_cluster') if context else None
>   File "/.../context.py", line 208, in load
>     return load_pickle(path)
>   File "/.../_misc.py", line 144, in load_pickle
>     return load(path)
>   File "/.../numpy_pickle.py", line 587, in load
>     obj = _unpickle(fobj, filename, mmap_mode)
>   File "/.../numpy_pickle.py", line 506, in _unpickle
>     obj = unpickler.load()
>   File "/.../pickle.py", line 1213, in load
>     dispatch[key[0]](self)
>   File "/.../pickle.py", line 1538, in load_stack_global
>     self.append(self.find_class(module, name))
>   File "/.../pickle.py", line 1580, in find_class
>     __import__(module, level=0)
> ModuleNotFoundError: No module named 'numpy._core'

410pfeliciano, 2024-06-23:
Try the following:
1-```pip install --force-reinstall numpy==1.26.4``` or 1.24.1
and Could you share the out put of the command ```pip show numpy``` ?

a-savoy, 2024-06-23:
I tried that, but due to other dependencies, the latest version I could install was 1.25.2. And I get the same error: `ModuleNotFoundError: No module named 'numpy._core'`. 

> $ pip show numpy
> Name: numpy
> Version: 1.25.2
> Summary: Fundamental package for array computing in Python
> Home-page: https://www.numpy.org
> Author: Travis E. Oliphant et al.
> Author-email: 
> License: BSD-3-Clause
> Location: /home/labadmin/anaconda3/envs/phy_env6/lib/python3.10/site-packages
> Requires: 
> Required-by: altair, bokeh, contourpy, deeplabcut, distinctipy, filterpy, h5py, hdbscan, hdmf, ibl-neuropixel, iblutil, imageio, imgaug, ipympl, isosplit5, matplotlib, mountainsort4, msgpack-numpy, neo, numba, numcodecs, numexpr, ONE-api, opencv-python, opt-einsum, pandas, patsy, phy, phylib, probeinterface, pyarrow, pyintan, pynwb, pyopencl, pyqtgraph, PyWavelets, scikit-image, scikit-learn, scipy, seaborn, spikeextractors, spikeinterface, statsmodels, tables, tensorboard, tensorflow, tensorpack, tifffile, tridesclous, zarr

zm711, 2024-06-24:
@a-savoy,

that error I actually know. It is because the NumPy limits aren't quite correct. So you installed some versions of packages necessary for NumPy 2.0 but when you downgraded to Numpy 1.25 or 1.26 etc it didn't downgrade the other packages. This is an unfortunate side effect of how pip resolves from the toml. So my advice would be to delete the conda environment all together and make a new one where you limit the numpy during the conda creation because if you install the wrong numpy during the conda install pip might not be able to fix it even if it downgrades the numpy itself. Does that make sense? 

But I think we are close.

zm711, 2024-06-24:
I also create #1289 if you want to test with the built in limit to numpy for making your environment.

a-savoy, 2024-06-24:
This makes sense, but what I did previously sounds like what you've described. I created a new Phy environment from a new .yml:

> name: phy_env6
> channels:
>   - conda-forge
>   - defaults
> dependencies:
>   - python=3.10
>   - pip
>   - git
>   - numpy=1.25.2
>   - matplotlib
>   - scipy
>   - h5py
>   - pyqt
>   - pyopengl
>   - pyqtwebengine
>   - pytest
>   - qtconsole
>   - requests
>   - responses
>   - traitlets
>   - dask
>   - cython
>   - pillow
>   - scikit-learn
>   - joblib
>   - pip:
>     - git+https://github.com/cortex-lab/phy.git

Should I manually set every package version to be compatible with the NumPy version?

zm711, 2024-06-25:
Oh, sorry I misunderstood. I thought you were just doing a pip install numpy=1.25 which would downgrade numpy but wouldn't necessarily fix the other packages. If you were installing from the beginning then one of the other packages must still be installing a too recent version. If you resize the window can you see where the pickle is being called from? ie can you give us a fuller file path in the debug window? Or share the log (note the log will have full file paths so edit that if you don't want all file paths displayed).

a-savoy, 2024-06-25:
**Update:** After restarting the computer and removing all but the most functional Phy environment (see package list below), I am able to load all Phy output from Kilosort4 and see the data as normal. When instead I try to load the output from SpikeInterface after running KS4 via the Docker, there is always one or more of the following problems: no waveforms, unusual waveforms, waveforms not shown according to the probe layout, or the numpy._core error. The same is true if I use a non-KS sorter sorter. 

So my Phy-related problem is solved enough for me to move forward. I'm giving up on SI for now, but I can open a new issue on the SI page if you'd like (or update the previous issue). 

Thanks for your help!

> # Name                    Version                   Build  Channel
> _libgcc_mutex             0.1                 conda_forge    conda-forge
> _openmp_mutex             4.5                       2_gnu    conda-forge
> alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
> asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
> attr                      2.5.1                h166bdaf_1    conda-forge
> aws-c-auth                0.7.22               h9137712_5    conda-forge
> aws-c-cal                 0.6.15               h88a6e22_0    conda-forge
> aws-c-common              0.9.19               h4ab18f5_0    conda-forge
> aws-c-compression         0.2.18               h83b837d_6    conda-forge
> aws-c-event-stream        0.4.2               h0cbf018_13    conda-forge
> aws-c-http                0.8.2                h360477d_2    conda-forge
> aws-c-io                  0.14.9               h2d549f9_2    conda-forge
> aws-c-mqtt                0.10.4               hf85b563_6    conda-forge
> aws-c-s3                  0.5.10               h679ed35_3    conda-forge
> aws-c-sdkutils            0.1.16               h83b837d_2    conda-forge
> aws-checksums             0.1.18               h83b837d_6    conda-forge
> aws-crt-cpp               0.26.12              h8bc9c4d_0    conda-forge
> aws-sdk-cpp               1.11.329             hf74b5d1_5    conda-forge
> bokeh                     3.4.1              pyhd8ed1ab_0    conda-forge
> brotli                    1.1.0                hd590300_1    conda-forge
> brotli-bin                1.1.0                hd590300_1    conda-forge
> brotli-python             1.1.0           py311hb755f60_1    conda-forge
> bzip2                     1.0.8                hd590300_5    conda-forge
> c-ares                    1.28.1               hd590300_0    conda-forge
> ca-certificates           2024.6.2             hbcca054_0    conda-forge
> cached-property           1.5.2                hd8ed1ab_1    conda-forge
> cached_property           1.5.2              pyha770c72_1    conda-forge
> cairo                     1.18.0               h3faef2a_0    conda-forge
> certifi                   2024.6.2           pyhd8ed1ab_0    conda-forge
> charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
> click                     8.1.7           unix_pyh707e725_0    conda-forge
> cloudpickle               3.0.0              pyhd8ed1ab_0    conda-forge
> colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
> colorcet                  3.1.0                    pypi_0    pypi
> comm                      0.2.2              pyhd8ed1ab_0    conda-forge
> contourpy                 1.2.1           py311h9547e67_0    conda-forge
> cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
> cython                    3.0.10          py311hb755f60_0    conda-forge
> cytoolz                   0.12.3          py311h459d7ec_0    conda-forge
> dask                      2024.6.0           pyhd8ed1ab_0    conda-forge
> dask-core                 2024.6.0           pyhd8ed1ab_0    conda-forge
> dask-expr                 1.1.3              pyhd8ed1ab_0    conda-forge
> dbus                      1.13.6               h5008d03_3    conda-forge
> debugpy                   1.8.1           py311hb755f60_0    conda-forge
> decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
> distributed               2024.6.0           pyhd8ed1ab_0    conda-forge
> exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
> executing                 2.0.1              pyhd8ed1ab_0    conda-forge
> expat                     2.6.2                h59595ed_0    conda-forge
> font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
> font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
> font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
> font-ttf-ubuntu           0.83                 h77eed37_2    conda-forge
> fontconfig                2.14.2               h14ed4e7_0    conda-forge
> fonts-conda-ecosystem     1                             0    conda-forge
> fonts-conda-forge         1                             0    conda-forge
> fonttools                 4.53.0          py311h331c9d8_0    conda-forge
> freetype                  2.12.1               h267a509_2    conda-forge
> fsspec                    2024.6.0           pyhff2d567_0    conda-forge
> gettext                   0.22.5               h59595ed_2    conda-forge
> gettext-tools             0.22.5               h59595ed_2    conda-forge
> gflags                    2.2.2             he1b5a44_1004    conda-forge
[... cut ...]
> xorg-recordproto          1.14.2            h7f98852_1002    conda-forge
> xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
> xorg-util-macros          1.19.3               h7f98852_0    conda-forge
> xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
> xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
> xorg-xproto               7.0.31            h7f98852_1007    conda-forge
> xyzservices               2024.6.0           pyhd8ed1ab_0    conda-forge

zm711, 2024-06-26:
> So my Phy-related problem is solved enough for me to move forward. I'm giving up on SI for now, but I can open a new issue on the SI page if you'd like (or update the previous issue).

Yeah based on this it seems like maybe some numpy stuff is has made our exporter not fully work. Feel free to reopen and update the old issue!

ronghao-zhang, 2024-09-10:
I also encounter the Segmentation fault (core dumped) issue when trying to input my data (generated with kilosort4) into Phy2 on Ubuntu 24.04.1 LTS. I was able to solve the problem using the `yml` file provided by @410pfeliciano. I think the key is to have an updated version of numpy and python 3.11 here ...

nataliekoh, 2024-11-08:
Just wanted to add that I used the yml file that @a-savoy provided to create a new environment and that completely fixed the issues I was getting with opening Kilosort4 outputs in Phy.

a-savoy, 2024-11-08:
@nataliekoh Wow, something I posted on Github actually helped someone?? This might be the happiest moment of my entire grad school experience! 🥹

naterenegar, 2025-01-08:
> The following installation([#1283 (comment)](https://github.com/cortex-lab/phy/issues/1283#issuecomment-2150453627)) solve my problem in Ubuntu 22.04:
> 
> name: phy2_test channels: -conda-forge -defaults dependencies: -python=3.11 -pip -git -numpy=1.26.4 -matplotlib -scipy -h5py -pyqt -pyopengl -pyqtwebengine -pytest -qtconsole -requests -responses -traitlets -dask -cython -pillow -scikit-learn -joblib -pip: -git+https://github.com/cortex-lab/phy.git

This env file fixed my issue, which was the same (segmentation fault during GUI building)