cortex-lab/phy #1291: Inexplicable OpenGL error loading datasets that previously opened without issue

a-savoy, 2024-07-01:
Last Friday I loaded a KS4-sorted dataset into Phy2 within a dedicated conda environment without any trouble. Today, I can neither load that dataset nor any other due to the error below. I have changed nothing in the environment, data, or on my computer. My best guess is that there was an automatic update to my system, but I'm running Ubuntu 22.04.2 without automatic updates activated. So this is a complete mystery to me. 

> $ phy template-gui params.py
> 12:10:19.577 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
> 12:10:21.100 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
> QGLXContext: Failed to create dummy context
> 0.00s - Debugger warning: It seems that frozen modules are being used, which may
> 0.00s - make the debugger miss breakpoints. Please pass -Xfrozen_modules=off
> 0.00s - to python to disable frozen modules.
> 0.00s - Note: Debugging will proceed. Set PYDEVD_DISABLE_FILE_VALIDATION=1 to disable this validation.
> Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
> Aborted (core dumped)
>

zm711, 2024-07-01:
@a-savoy, this is the third example of this happening. (2x on Windows--which has more aggressive  updating) where Phy was working and then suddenly stopped without any user input. I'm still trying to figure out why... Thanks for the report, nothing has been changed in the opengl section of the code, so I honestly don't know how this could happen. Do you want to create an environment and try to install with just pip from #1289 ( I added a numpy limit to the requirements.txt) and see if that fixes it. The other folks couldn't get the conda to work anymore (so I wonder if conda-forge has a point release that is breaking stuff).

a-savoy, 2024-07-01:
Good news: simply restarting the computer seems to have solve the problem. This is the second time for me that a restart has magically resolved a Phy issue.

zm711, 2024-07-02:
Well let the record show that restarting the computer helps! :) 

Unfortunately I know python more than I know any of the renderer stuff, so the opengl/qt is unknown to me. But since rendering communicates with OS there is probably some sort of cache issue etc and so a restart cleans that up. But I don't have a good reason.

jenliujenliujen, 2025-03-28:
Hi I'm using ubuntu22.04.3 and looks like I encountered a similar issue but with *libGL error: MESA-LOADER*, I've also tried restarting but does not solve the problem, is there any update for this issue? Thanks!

phy template-gui params.py

> 13:22:38.499 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
> 13:22:39.680 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
> 13:22:41.184 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: swrast
> WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
> QGLXContext: Failed to create dummy context
> Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
> Aborted (core dumped)

Here's the debug:
phy template-gui params.py --debug

> 13:25:11.334 [D] __init__:68          Start capturing exceptions.
> 13:25:15.362 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
> 13:25:15.639 [D] model:619            Loading spike clusters.
> 13:25:16.576 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
> 13:25:16.882 [D] model:569            No channel shank file found.
> 13:25:16.892 [D] model:692            Loading templates.
> 13:25:18.076 [D] model:724            Templates are dense.
> 13:25:18.098 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 13:25:18.098 [D] model:730            Loading the whitening matrix.
> 13:25:18.138 [D] model:737            Loading the inverse of the whitening matrix.
> 13:25:18.216 [D] model:766            Loading features.
> 13:25:18.218 [D] model:803            Loading template features.
> 13:25:18.247 [D] model:504            Load `metrics.csv`.
> 13:25:18.267 [D] model:504            Load `cluster_KSLabel.tsv`.
> 13:25:18.282 [D] model:504            Load `cluster_Amplitude.tsv`.
> 13:25:18.299 [D] model:504            Load `cluster_group.tsv`.
> 13:25:18.314 [D] model:504            Load `cluster_ContamPct.tsv`.
> libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
> libGL error: failed to load driver: swrast
> 13:25:18.560 [D] context:100          Initialize joblib cache dir at `/media/nerfceph01_farrowlabwip2024/Data/02285/neuropixels/498L/20240705/ephys/sorted_IBL_full_KS3/jen_si2_02285_498L_20240705_IBL_full_KS3_16-01-2025/sorting_kilosort3/sorter_output/.phy`.
> 13:25:18.565 [D] context:101          Reducing the size of the cache if needed.
> 13:25:18.721 [D] base:102             Add filter `high_pass`.
> 13:25:18.727 [D] config:31            Load config file `/home/jenliu/.phy/phy_config.py`.
> 13:25:18.733 [D] plugin:146           Loading 0 plugins.
> 13:25:18.780 [D] context:209          The file `path` doesn't exist.
> 13:25:20.389 [D] context:185          Save data to `path`.
> 13:25:21.452 [D] gui:463              Creating GUI.
> 13:25:21.507 [D] state:46             Load /home/jenliu/.phy/TemplateGUI/state.json for GUIState.
> WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
> QGLXContext: Failed to create dummy context
> 13:25:22.364 [D] gui:718              Add view ClusterView to GUI.
> 13:25:22.376 [D] gui:718              Add view SimilarityView to GUI.
> 13:25:22.429 [D] gui:718              Add view WaveformView to GUI.
> 13:25:22.442 [D] base:337             Set state for WaveformView.
> 13:25:22.467 [D] gui:718              Add view WaveformView to GUI.
> 13:25:22.474 [D] base:337             Set state for WaveformView (1).
> 13:25:22.499 [D] gui:718              Add view WaveformView to GUI.
> 13:25:22.506 [D] base:337             Set state for WaveformView (2).
> 13:25:22.527 [D] gui:718              Add view CorrelogramView to GUI.
> 13:25:22.534 [D] base:337             Set state for CorrelogramView.
> 13:25:22.571 [D] gui:718              Add view ISIView to GUI.
> 13:25:22.578 [D] base:337             Set state for ISIView.
> 13:25:22.619 [D] gui:718              Add view AmplitudeView to GUI.
> 13:25:22.626 [D] base:337             Set state for AmplitudeView.
> 13:25:22.731 [D] gui:718              Add view ProbeView to GUI.
> 13:25:22.737 [D] base:337             Set state for ProbeView.
> 13:25:22.765 [D] gui:718              Add view ClusterScatterView to GUI.
> 13:25:22.772 [D] base:337             Set state for ClusterScatterView.
> 13:25:22.777 [D] base:494             Set color scheme to random.
> 13:25:22.784 [D] base:1626            Adding default color schemes to ClusterScatterView.
> Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
> Aborted (core dumped)
>