cortex-lab/phy #1292: ModuleNotFoundError: No module named 'numpy._core' on Ubuntu 22.04

Idavr, 2024-07-03:
I have downloaded phy according to all instructions (tried both directly with git and by environment.yml files) and gone over multiple tweaks on other opened and closed issues (like installing with different versions of python and numpy) but to no avail. The data I am trying to look at has been spike sorted with Kilosort 4 through their terminal commands (no GUI). No matter what I do I still get this error:

(phy2) idavalik@ida-HP-Z4-G5-Workstation:~/Kilosort/test_data/kilosort4$ phy template-gui params.py
13:54:26.548 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
13:54:26.755 [E] __init__:62          An error has occurred (ModuleNotFoundError): No module named 'numpy._core'
Traceback (most recent call last):
  File "/home/idavalik/anaconda3/envs/phy2/bin/phy", line 8, in <module>
    sys.exit(phycli())
             ^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/template/gui.py", line 217, in template_gui
    controller = TemplateController(model=model, dir_path=dir_path, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/base.py", line 470, in __init__
    super(TemplateMixin, self).__init__(*args, **kwargs)
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/base.py", line 926, in __init__
    self._set_supervisor()
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/template/gui.py", line 95, in _set_supervisor
    super(TemplateController, self)._set_supervisor()
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/base.py", line 1010, in _set_supervisor
    supervisor = Supervisor(
                 ^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/cluster/supervisor.py", line 637, in __init__
    spc = context.load('spikes_per_cluster') if context else None
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/utils/context.py", line 208, in load
    return load_pickle(path)
           ^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/phylib/utils/_misc.py", line 144, in load_pickle
    return load(path)
           ^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/joblib/numpy_pickle.py", line 658, in load
    obj = _unpickle(fobj, filename, mmap_mode)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/site-packages/joblib/numpy_pickle.py", line 577, in _unpickle
    obj = unpickler.load()
          ^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/pickle.py", line 1213, in load
    dispatch[key[0]](self)
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/pickle.py", line 1538, in load_stack_global
    self.append(self.find_class(module, name))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/idavalik/anaconda3/envs/phy2/lib/python3.11/pickle.py", line 1580, in find_class
    __import__(module, level=0)
ModuleNotFoundError: No module named 'numpy._core'

This is my conda package list:

# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
[... 292 lines cut ...]
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxtst              1.2.3             h7f98852_1002    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-recordproto          1.14.2            h7f98852_1002    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-util-macros          1.19.3               h7f98852_0    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xf86vidmodeproto     2.3.1             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xyzservices               2024.6.0           pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.5                h75354e8_4    conda-forge
zict                      3.0.0              pyhd8ed1ab_0    conda-forge
zipp                      3.19.2             pyhd8ed1ab_0    conda-forge
zlib                      1.3.1                h4ab18f5_1    conda-forge
zstandard                 0.22.0          py311hb6f056b_1    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge

As mentioned in the title, I am running on an Ubuntu 22.04.

Any help, insights, further steps for troubleshooting are all welcome!

zm711, 2024-07-03:
Typically this has been due to mixing conda + pip which means the packages aren't quite playing well together. Could you try:

first clone the repo with git (or download the zip if not using git). Then edit the `requirements.txt` file such that numpy<2.0

```python
conda create -n phy2_test python=3.11
cd phy
pip install -e .
```

And let me know if that works. We may need to add in a step where we also update phylib.

Idavr, 2024-07-03:
I see! I tried following your instructions and now I got a different error:

14:15:27.361 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
14:15:30.331 [W] gui:671              Could not create view TemplateFeatureView.
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)

Idavr, 2024-07-04:
The restart worked! Thank you so much!

zm711, 2024-07-10:
Great!

closed 2024-07-10 by zm711

Abbsalehi, 2025-03-31:
see #14160