cortex-lab/phy #1294: ImportError: DLL load failed while importing _imaging

H-Jiyoung, 2024-07-04:
Hi all, 

Recently, I tried to reinstall phy2, but I cannot execute GUI because of 'Import Error'.
How can I solve this problem...?
Attached picture showed the error message. 
Thank you in advance.

Python ver. = 3.11.9
[image]

zm711, 2024-07-05:
Could you do me a favor and translate the last bit of Korean. It says `ImportError: DLL load failed while importing _imaging:...`

I just want to make sure that's not important info.

Second, how did you install phy?

H-Jiyoung, 2024-07-06:
Hi,
Thank you for your reply.
I strictly followed all instructions in the installation part of phy GitHub. The Korean part means "The specified module was not found". Fortunately, I solved the "Import Error" by typing `pip install -U Pillow` but there is still the same issue as #1290. Is there any method to open 'TemplateFeature View'?
The error message is attached below.

[image]

zm711, 2024-07-10:
Sorry for the delay. Great that you got your immediate issue fixed. Which KS are you using. Or are you using Spikeinterface. If you are using KS3 they didn't fully implement phy functionality. So you could do

```
phy extract-waveforms params.py
```

but this causes a "frozen" state so a lot of people complain about using this. If you go to do a split often the numbers aren't what people expect because this only takes a subsample rather than the real data so it is prone to some pretty wild sampling biases. But if using KS3 it is the only way you can use some version of the feature view.

SpikeInterface has an export_to_phy that can provide most of the phy views even with KS3 sorted data.

zm711, 2024-08-08:
I'm going to close this for lack of responses. Just open a new issue if something else comes up.

closed 2024-08-08 by zm711