cortex-lab/phy #1299: Installed Phy according to instructions, cannot open GUI

AidanFishenden, 2024-08-07:
I installed Phy exactly according to the instructions, and upon trying to open a kilosort file with  it encountered the following error. 

Usage: phy template-gui [OPTIONS] PARAMS_PATH
Try 'phy template-gui --help' for help.
Error: Invalid value for 'PARAMS_PATH': Path 'params.phy' does not exist.

Trying the help option gives 
  --clear-state / --no-clear-state
                                  Clear the GUI state in `~/.phy/` and in
                                  `.phy`.
  --clear-cache / --no-clear-cache
                                  Clear the .phy cache in the data directory.
  --help                          Show this message and exit.

but trying any of these options results in an error stating it is not recognized as an internal or external command, operable program or batch file. 

I am very new to python/anaconda so I am not sure if this matters, but I installed everything with the phy2 environment activated.

zm711, 2024-08-07:
What did you exactly type in?

For example if my kilosort data is in a folder called my_data I could cd there like so 

```
$ cd my_data
$ conda activate phy2
$ phy template-gui params.py
```

So first I go to the folder that has the file called `params.py` in it. Then I make sure phy2 is active. Then I use `phy` with the command `template-gui` command and point that command to the `params.py` file.

And no worries! We all have to start somewhere :)

AidanFishenden, 2024-08-07:
Sorry for the typo in my original post, that is not the source of my error in script. This is exactly what I am typing

cd E:\Dataforkilosort\OneDrive_1_6-15-2023\1_ftc\kilosort4
Conda activate phy2
phy tenplate-gui params.py

I checked, the params.py is in my kilosort folder.

zm711, 2024-08-07:
if you open the params.py file in the folder what does it say inside? it should have lines like
`dat_path=r'path'`
etc.

And is your cd command working? Which shell are you using on Windows. For anaconda prompt you may need to do cd/d in order to properly switch drives.

closed 2024-08-07 by AidanFishenden

zm711, 2024-08-07:
No worries. Almost all instructions are for Linux so some of the small things for windows can be missed. let me give you one more huge tip for working on windows in python. Anytime you need to give a path instead of 

```
my_path = "path\to\my\data"
```

do 
```
my_path = r"path\to\my\data"
```

This prevents spurious escaping in file paths. It will save you some crazy issues in the future :)

AidanFishenden, 2024-08-07:
I will make a note of that for the future, thank you!