cortex-lab/phy #1300: [HELP] Time of the waveforms

EMerinoM14, 2024-08-07:
Hello, I have a question, I think it is easy. 

How much time is represented in the gray line?
I guess we are in millisecond scales but how many milliseconds are represented there?

Thanks

[image]

zm711, 2024-08-07:
Not quite milliseconds directly. The axis is samples, so it depends on the sample rate of your recording. The two most common amounts are either 61 samples or 81 samples.  So the standard 30,000 sampling rate that many systems use with mean 

61 / 30_000 ~ 0.002 seconds or about 2 ms.

zm711, 2024-09-06:
Let me know if you need more explanation.

closed 2024-09-06 by zm711