cortex-lab/phy #1301: Not seeing raw waveforms in the WaveformView of phy (but template waveforms appear)

chrisan97, 2024-08-08:
Hi everyone, I have a weird problem that started happening to me maybe a week ago or so. In the WaveformView on phy, I am able to see template waveforms like: 
[image]

However, when I switch to see the raw waveforms, I don't see any waveforms: 
[image]

I am currently using KS2.5 and recently I switched from using the GUI to using the matlab script to run KS. Have anyone seem this issue before? What would be the corresponding KS output that allows you to load these raw waveforms into phy? I'm wondering if somehow one of the files do not get generated properly when I use the KS script vs the KS2.5 GUI.

Thanks!

zm711, 2024-08-08:
There was the spike hole issue in KS. you may need to make sure the script is working okay. This is just using the spike times that kilosort saves (`spike_times.npy`) in order to load the waveforms from the raw data. So you could make your own and double check that the spike times from the script are correct for these channels. If they end up being correct then maybe there is a gui issue, but if they are not correct you'd have to let KS people know.

chrisan97, 2024-08-08:
I can definitely load the spike times into my custom script and view the raw waveforms that way, it's just a weird bug that I started noticing once I switched over to the script version of KS2.5. I believe I have updated KS2.5 to the newest version to fix the spike hole issue. Seems like it's a gui issue? I can also try rerunning the exact same session with the GUI version of KS and see if the issue persists.

zm711, 2024-08-08:
Yeah I would try that. and see what happens!

chrisan97, 2024-08-08:
Yes, as suspected the script version of KS2.5 has issues w.r.t generating the raw waveforms in phy. There are no differences in file size/file number that are generated with KS2.5. I ran `phy template-gui params.py --debug` and compared the output but they seemed identical to me. Any guesses on what might be causing this? Or should I direct this question to the Kilosort folks?

Thanks!

zm711, 2024-08-08:
I think the spike holes issue required some careful checking of batch size. So if I were you I would check the batch sizes between the gui code and the script code and see if the batch sizes match. If they don't then that's the issue (if they don't match it would cause a shift such that the raw waveforms would be a batch off). If it's not that then unfortunately you'd have to check with them.

chrisan97, 2024-08-08:
Thanks a ton! Will investigate.

closed 2024-08-08 by chrisan97