cortex-lab/phy #1302: [HELP] Import Correlogram

EMerinoM14, 2024-08-18:
Hi,

I would like to calculate the rhythmicity index of the neurons and for this I would like to know if it is possible to extract the correlogram of each of the clusters in Matlab.

If this is not possible, could someone give me a hint on how to calculate it?

Best

JoeZiminski, 2024-08-19:
Hi @EMerinoM14,

I don't think it is possible to directly extract the correlograms from the phy GUI. However, it should be possible to calculate the correlograms using the internal code and save the outputs to something you can read in MATLAB (e.g. a `.csv`).

The code for the computation of the correlogram is [here](https://github.com/cortex-lab/phylib/blob/de4993e2f708f24eb98a02d6373b28b1277f2815/phylib/stats/ccg.py#L79). It takes only the spike times and unit labels, which I think you can load directly from the sorter output (e.g. `spike_times.npy` and `spike_clusters.npy`) with `np.load`. You might need to check the format of the data before passing it to the correlogram, but you can check your outputs look correct against the phy gui. Then you can save to your preferred output filetype in Python and load into MATLAB.

Alternatively, there are a few toolboxes (e.g. [1](https://github.com/cortex-lab/spikes), [2](extracellular ephys)) for working directly in MATLAB, that should be able to take kilosort outputs directly, but I have never used them myself. Let me know if you have any questions.

zm711, 2024-09-06:
Thanks @JoeZiminski.

I'm closing this since this is a great response for matlab options. If you want to do more python spikeinterface is a great option!

closed 2024-09-06 by zm711