cortex-lab/phy #1320: Segmentation fault using  phy template-gui params.py

FabianMunozSilva, 2024-12-17:
Hi Evereyone, 
I just started with Kilosort, and it is working but when I run 
phy template-gui params.py --debug
14:44:52.938 [D] __init__:68          Start capturing exceptions.
14:44:52.963 [D] model:619            Loading spike clusters.
14:44:52.983 [D] model:692            Loading templates.
14:44:52.992 [D] model:724            Templates are dense.
14:44:52.992 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
14:44:52.992 [D] model:730            Loading the whitening matrix.
14:44:52.992 [D] model:737            Loading the inverse of the whitening matrix.
14:44:52.993 [D] model:766            Loading features.
14:44:52.994 [D] model:781            Features are sparse.
14:44:52.994 [D] model:803            Loading template features.
14:44:52.995 [D] model:530            Load spike_detection_templates.npy.
14:44:52.996 [D] model:530            Load spike_positions.npy.
14:44:52.996 [D] model:504            Load `cluster_Amplitude.tsv`.
14:44:52.997 [D] model:504            Load `cluster_group.tsv`.
14:44:52.998 [D] model:504            Load `cluster_ContamPct.tsv`.
14:44:52.999 [D] model:504            Load `cluster_KSLabel.tsv`.
14:44:53.141 [D] context:100          Initialize joblib cache dir at `/home/vpflab/Documents/K4/kilosort4/.phy`.
14:44:53.141 [D] context:101          Reducing the size of the cache if needed.
14:44:53.142 [D] base:102             Add filter `high_pass`.
14:44:53.142 [D] config:31            Load config file `/home/vpflab/.phy/phy_config.py`.
14:44:53.142 [D] plugin:146           Loading 0 plugins.
14:44:53.143 [D] context:209          The file `/home/vpflab/Documents/K4/kilosort4/.phy/new_cluster_id.pkl` doesn't exist.
14:44:53.158 [D] context:185          Save data to `/home/vpflab/Documents/K4/kilosort4/.phy/spikes_per_cluster.pkl`.
14:44:53.190 [D] gui:463              Creating GUI.
14:44:53.195 [D] state:46             Load /home/vpflab/.phy/TemplateGUI/state.json for GUIState.
Segmentation fault (core dumped)
I’ve reviewed previous posts but haven’t found a solution to my problem. I can run phy trace-gui params.py, so I suspect the issue is related to the graphical interface. However, I’m unsure how to resolve it.
For reference, I’m using Ubuntu 24.04.1 LTS.
Any guidance would be greatly appreciated!
Thanks in advance
Fabian

FabianMunozSilva, 2024-12-17:
I change the python from 3.9 to 3.12 in environment.yml installation and I can see the GUI but I can not see the traces I have this message 
15:18:46.798 [W] gui:671              Could not create view TemplateFeatureView.
dict_keys([<class 'numpy.ndarray'>, <class 'bytes'>, <class 'OpenGL.converters.c_int_Array_1'>])

ilsedippenaar, 2024-12-20:
Hi! I'm running into a similar problem, also on Ubuntu 24.04.1. I ran these installation steps:

```bash
conda create -n phy2 -y python=3.11
conda activate phy2
conda install -c conda-forge libstdcxx-ng  # needed to allow MESA-LOADER to find iris and swrast shared libraries
pip install git+https://github.com/cortex-lab/phy.git
```
I also needed to update conda to >24 to avoid needing to mess with LD_PRELOAD in order to pull a more recent C library for the `iris_dri.so` and `swrast_dri.so` binaries. After doing all of that, I experience what's described above (Phy launches, but no traces displayed). However, in my case, Phy throws a lot of `Exception in paintGL: Attempt to retrieve context when no valid context` errors.

If I try installing with `conda env update environment.yml --prune`, I get a host of version errors or missing package errors (e.g., libiconv, libicuuc) with Glib version mismatches as well.

Weirdly, it _does_ work fine on a different computer running Ubuntu 24.04.1 but that computer has a GPU. In that case, I didn't need to update `libstdcxx-ng`, but I did need to install directly from environment.yml. In case it's helpful, I've copied the full frozen environment from that computer below, but there are still plenty of version conflicts when I install on the first computer.
<details>
<summary>environment.yml</summary>

```yml
name: phy2
channels:
  - conda-forge
  - defaults
dependencies:
  - _libgcc_mutex=0.1=conda_forge
  - _openmp_mutex=4.5=2_gnu
  - alsa-lib=1.2.13=hb9d3cd8_0
  - asttokens=3.0.0=pyhd8ed1ab_1
  - attr=2.5.1=h166bdaf_1
  - aws-c-auth=0.8.0=hb921021_15
  - aws-c-cal=0.8.1=h1a47875_3
  - aws-c-common=0.10.6=hb9d3cd8_0
  - aws-c-compression=0.3.0=h4e1184b_5
  - aws-c-event-stream=0.5.0=h7959bf6_11
  - aws-c-http=0.9.2=hefd7a92_4
  - aws-c-io=0.15.3=h831e299_5
  - aws-c-mqtt=0.11.0=h11f4f37_12
  - aws-c-s3=0.7.7=hf454442_0
  - aws-c-sdkutils=0.2.1=h4e1184b_4
  - aws-checksums=0.2.2=h4e1184b_4
  - aws-crt-cpp=0.29.7=hd92328a_7
  - aws-sdk-cpp=1.11.458=hc430e4a_4
  - azure-core-cpp=1.14.0=h5cfcd09_0
  - azure-identity-cpp=1.10.0=h113e628_0
  - azure-storage-blobs-cpp=12.13.0=h3cf044e_1
  - azure-storage-common-cpp=12.8.0=h736e048_1
  - azure-storage-files-datalake-cpp=12.12.0=ha633028_1
  - bokeh=3.6.2=pyhd8ed1ab_1
  - brotli=1.1.0=hb9d3cd8_2
  - brotli-bin=1.1.0=hb9d3cd8_2
  - brotli-python=1.1.0=py311hfdbb021_2
  - bzip2=1.0.8=h5eee18b_6
  - c-ares=1.34.4=hb9d3cd8_0
  - ca-certificates=2024.12.14=hbcca054_0
  - cached-property=1.5.2=hd8ed1ab_1
  - cached_property=1.5.2=pyha770c72_1
  - cairo=1.18.2=h3394656_1
  - certifi=2024.12.14=pyhd8ed1ab_0
  - cffi=1.17.1=py311hf29c0ef_0
  - charset-normalizer=3.4.0=pyhd8ed1ab_1
  - click=8.1.7=unix_pyh707e725_1
  - cloudpickle=3.1.0=pyhd8ed1ab_1
  - colorama=0.4.6=pyhd8ed1ab_1
  - comm=0.2.2=pyhd8ed1ab_1
  - contourpy=1.3.1=py311hd18a35c_0
  - cycler=0.12.1=pyhd8ed1ab_1
  - cyrus-sasl=2.1.27=h54b06d7_7
  - cython=3.0.11=py311h55d416d_3
  - cytoolz=1.0.1=py311h9ecbd09_0
  - dask=2024.12.1=pyhd8ed1ab_0
  - dask-core=2024.12.1=pyhd8ed1ab_0
  - dask-expr=1.1.21=pyhd8ed1ab_0
[... 252 lines cut ...]
  - xorg-libxxf86vm=1.1.6=hb9d3cd8_0
  - xorg-xf86vidmodeproto=2.3.1=hb9d3cd8_1005
  - xyzservices=2024.9.0=pyhd8ed1ab_1
  - yaml=0.2.5=h7f98852_2
  - zeromq=4.3.5=h3b0a872_7
  - zict=3.0.0=pyhd8ed1ab_1
  - zipp=3.21.0=pyhd8ed1ab_1
  - zstandard=0.23.0=py311hbc35293_1
  - zstd=1.5.6=ha6fb4c9_0
  - pip:
      - colorcet==3.1.0
      - ghp-import==2.1.0
      - markdown==3.7
      - mergedeep==1.3.4
      - mkdocs==1.6.1
      - mkdocs-get-deps==0.2.0
      - mtscomp==1.0.2
      - pathspec==0.12.1
      - phy==2.0b5
      - phylib==2.6.0
      - pyyaml-env-tag==0.1
      - tqdm==4.67.1
      - watchdog==6.0.0
```
</details>