cortex-lab/phy #1330: Only able to display ClusterView and SimilarityView - Python 3.12 (used new .yaml #1283) + Intel(R) UHD Graphics 770

cknox0119, 2025-02-13:
I am unable to display any of the data outside of ClusterView and SimilarityView, but my labmate was able to have the GUI run properly off of my Kilosort sorter_output file. He is using Python 3.11.5 and I am using Python 3.12.7. Because of these differences I completely uninstalled the initial setup of Phy2 and tried to startup the new environment from the .yaml file from issue #1283. It still fails to display the analysis windows. 

Additionally I checked and confirmed my graphics card (Intel(R) UHD Graphics 770 is OpenGL compatible. I even tried changing to my NVDIA which also failed to fix the GUI display.

cknox0119, 2025-02-13:
Resolved :)

MarcoUniPr, 2025-03-25:
how did you resolve?
Thanks