cortex-lab/phy #1335: Waveform view issue with x11 session

Jacob-Worrell, 2025-03-19:
I am currently trying to get a new workflow set up for Phy, and encountering an issue with the waveform view.

My use case is a little bit unusual; I connect via ThinLinc to a research computing virtual desktop environment, and then use ssh with x11 forwarding to access a different account on the same research computing cluster, from which I can then request an interactive job on a compute node, and launch my Phy session.

The issue I am encountering in doing it this way is that I can no longer toggle "type" in the waveform view window. This happens whether I use the W hotkey or the waveform view drop-down menu. If I inspect the hotkeys, "W" is still assigned to toggle type.

Just in case it was still somehow a hotkey issue, though, I did try <phy template-gui params.py --clear-state> in order to reset hotkey and window preferences.

Jacob-Worrell, 2025-03-20:
closing, turned out to be an error with path specification on my part.

closed 2025-03-20 by Jacob-Worrell

mandarmp, 2025-05-25:
Hello @Jacob-Worrell ,

I am trying to run the phy on perlmutter NERSC, but I get this error, I am using ThinLinc that ssh to perlmutter I suppose, I get this error 

208/M07038/Network/000027/well000/kilosort2__rec0000/sorter_output> phy template-gui params.py --debug
20:39:34.477 [D] __init__:68          Start capturing exceptions.
20:39:34.735 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
20:39:34.886 [D] model:619            Loading spike clusters.
20:39:35.040 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
20:39:35.188 [D] model:569            No channel shank file found.
20:39:35.189 [D] model:692            Loading templates.
20:39:35.417 [D] model:724            Templates are dense.
20:39:35.421 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
20:39:35.422 [D] model:730            Loading the whitening matrix.
20:39:35.425 [D] model:737            Loading the inverse of the whitening matrix.
20:39:35.434 [D] model:766            Loading features.
20:39:35.440 [D] model:781            Features are sparse.
20:39:35.441 [D] model:803            Loading template features.
20:39:35.445 [D] model:813            Template features are sparse.
20:39:35.446 [D] model:504            Load `cluster_group.tsv`.
20:39:35.449 [D] model:504            Load `cluster_ContamPct.tsv`.
20:39:35.451 [D] model:504            Load `cluster_KSLabel.tsv`.
20:39:35.454 [D] model:504            Load `cluster_Amplitude.tsv`.
20:39:35.547 [D] context:100          Initialize joblib cache dir at `/pscratch/sd/m/mpatil1/MEA_Analysis/AnalyzedData/CDKL5_R59X_SingleBrainPerChip_2/CDKL5-R59X_SingleBrainPerChip_11252024_PS/241208/M07038/Network/000027/well000/kilosort2__rec0000/sorter_output/.phy`.
20:39:35.547 [D] context:101          Reducing the size of the cache if needed.
20:39:35.554 [D] base:102             Add filter `high_pass`.
20:39:35.555 [D] config:31            Load config file `/global/homes/m/mpatil1/.phy/phy_config.py`.
20:39:35.555 [D] plugin:146           Loading 0 plugins.
20:39:35.557 [D] context:209          The file `/pscratch/sd/m/mpatil1/MEA_Analysis/AnalyzedData/CDKL5_R59X_SingleBrainPerChip_2/CDKL5-R59X_SingleBrainPerChip_11252024_PS/241208/M07038/Network/000027/well000/kilosort2__rec0000/sorter_output/.phy/new_cluster_id.pkl` doesn't exist.
20:39:35.664 [D] context:185          Save data to `/pscratch/sd/m/mpatil1/MEA_Analysis/AnalyzedData/CDKL5_R59X_SingleBrainPerChip_2/CDKL5-R59X_SingleBrainPerChip_11252024_PS/241208/M07038/Network/000027/well000/kilosort2__rec0000/sorter_output/.phy/spikes_per_cluster.pkl`.
20:39:35.807 [D] gui:463              Creating GUI.
20:39:35.817 [D] state:46             Load /global/homes/m/mpatil1/.phy/TemplateGUI/state.json for GUIState.
QGLXContext: Failed to create dummy context
20:39:39.640 [D] gui:718              Add view ClusterView to GUI.
20:39:39.647 [D] gui:718              Add view SimilarityView to GUI.
20:39:39.686 [D] gui:718              Add view WaveformView to GUI.
20:39:39.688 [D] base:337             Set state for WaveformView.
20:39:39.701 [D] gui:718              Add view CorrelogramView to GUI.
20:39:39.702 [D] base:337             Set state for CorrelogramView.
20:39:39.729 [D] gui:718              Add view ISIView to GUI.
20:39:39.731 [D] base:337             Set state for ISIView.
20:39:39.743 [D] gui:718              Add view FeatureView to GUI.
20:39:39.744 [D] base:337             Set state for FeatureView.
20:39:39.766 [D] gui:718              Add view AmplitudeView to GUI.
20:39:39.767 [D] base:337             Set state for AmplitudeView.
20:39:39.789 [D] gui:718              Add view FiringRateView to GUI.
20:39:39.790 [D] base:337             Set state for FiringRateView.
20:39:40.710 [D] gui:718              Add view TraceView to GUI.
20:39:40.712 [D] base:337             Set state for TraceView.
20:39:40.712 [D] base:494             Set color scheme to depth.
20:39:40.993 [D] base:1626            Adding default color schemes to TraceView.
20:39:42.796 [D] gui:718              Add view ProbeView to GUI.
20:39:42.799 [D] base:337             Set state for ProbeView.
20:39:42.817 [D] gui:718              Add view TemplateFeatureView to GUI.
20:39:42.819 [D] base:337             Set state for TemplateFeatureView.
20:39:42.838 [D] gui:718              Add view ClusterScatterView to GUI.
20:39:42.840 [D] base:337             Set state for ClusterScatterView.
20:39:42.840 [D] base:494             Set color scheme to firing_rate.
20:39:42.841 [D] base:1626            Adding default color schemes to ClusterScatterView.
20:39:42.906 [D] gui:519              Load the geometry state.
QOpenGLWindow::beginPaint: Failed to create context
QOpenGLWindow::beginPaint: Failed to make context current
QOpenGLFunctions created with non-current context
Segmentation fault

do you have any idea?