cortex-lab/phy #1339: Problem to start phy-gui

belalipaiva, 2025-04-24:
Hello , I am using the phy2 and since last week the application stopped working showing the error bellow. 
I've already deleted the enviroment and also its associated folders and reinstalled the enviroment a couple of times, folowing strictly the steps and also reinstall with the enviroment.yml file but noting works.

Many thanks, in advance for any help you could provide: 

Error: 

Info: I was working in the very same session before and then I closed the application, and when I tried to open again it did not work for any of the sessions. 

(phy2) E:\CHEESEBOARDMAZE\Rat188\Task\Rat188-09032024\kilosort4>phy template-gui params.py --debug
11:22:25.792 [D] __init__:68          Start capturing exceptions.
11:22:26.831 [E] __init__:62          An error has occurred (ValueError): The spike times must be increasing.
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\izabela.lima\.conda\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
             ^^^^^^^^
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\click\core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\click\core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\click\core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\click\core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\click\core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\izabela.lima\.conda\envs\phy2\Lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui
    model = load_model(params_path)
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\AppData\Roaming\Python\Python311\site-packages\phylib\io\model.py", line 1440, in load_model
    return TemplateModel(**get_template_params(params_path))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\izabela.lima\AppData\Roaming\Python\Python311\site-packages\phylib\io\model.py", line 339, in __init__
    self._load_data()
  File "C:\Users\izabela.lima\AppData\Roaming\Python\Python311\site-packages\phylib\io\model.py", line 353, in _load_data
    raise ValueError("The spike times must be increasing.")
ValueError: The spike times must be increasing.

QWidget: Must construct a QApplication before a QWidget

darianyao, 2025-06-03:
Hi, I am also encounter a similar problem, could you please teach me how to fix it?  The error it showed displays bellow:
21:18:18.138 [E] __init__:62          An error has occurred (TypeError): Memory.__init__() got an unexpected keyword argument 'bytes_limit'
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\20155\anaconda3\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
             ^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1442, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1363, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1830, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1226, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 794, in invoke
    return callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\click\decorators.py", line 34, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-packages\phy\apps\template\gui.py", line 217, in template_gui
    controller = TemplateController(model=model, dir_path=dir_path, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\20155\anaconda3\envs\phy2\Lib\site-pa

bitnajooo, 2025-08-29:
Here's what we did. 
In the phy2 environment,

````
pip uninstall joblib
pip install joblib==1.4.2
````